libimobiledevice/libimobiledevice #169: iOS 8 issue with lockdownd_get_value.

coolstar, 2015-01-05:
On iOS 8.1.1 (maybe also other iOS 8 versions), lockdownd_get_value only works properly the first time. Calling it a second time (even for a different value) always returns a return value of 1 and never gets the value properly. It works as intended on iOS 5.1.1, 6.1.3 & 7.1.2

nikias, 2015-01-05:
Not sure what you're doing exactly, but I can't reproduce it. It would be helpful to see the code. I tried this small program and it works perfectly with iOS 8: http://pastie.org/private/oycyr1sijyrymqwjwolxqw

coolstar, 2015-01-09:
This is the code I used previously. https://ghostbin.com/paste/ptv3q

nikias, 2015-01-09:
I can't spot any mistake in your code, have you tried mine? Please compile libimobiledevice with debug output and add a line idevice_set_debug_level(1); at the beginning of the code to see if the device actually returns something useful or not.

nikias, 2015-01-24:
Any new observations? Otherwise I'll have to close the ticket.

coolstar, 2015-01-24:
Still notice the issue here. I'm running the libimobiledevice available from Ubuntu 14.04's repositories. Are you running a different version? Might be it

nikias, 2015-01-24:
I'm using latest git head of course. Still that should work, what about running with debug output to see if the device actually responds?

coolstar, 2015-01-24:
how would I enable debug output from libimobiledevice?

nikias, 2015-01-24:
libimobiledevice has to be compiled with debug output, passing --enable-debug-code to autogen.sh would enable it. On ubuntu there should be a libimobiledevice-dbg package that should have the debug output enabled. Then add a line idevice_set_debug_level(1); at the beginning of your code and you can see the communication.