libimobiledevice/libimobiledevice #170: 32bit architecture issue

girishk55, 2015-01-16:
I compile this library on my mac which have OSX 10.9.5 version. using following commands. 

sudo ./configure
sudo make && sudo make install

and integrate compiled libimobiledevice.4.dylib file in my project. If i run the same code on 64 bit ($(ARCHS_STANDARD)) architecture it is working fine. but if i change the architecture  32bit ($(ARCHS_STANDARD_32_64_BIT)) it producing following error.

ld: warning: ignoring file /Users/admin/Sumit_Projects/Iphone/SamartRecovery/Smart iPhoneRecovery ProAmit/libimobiledevice.4.dylib, file was built for x86_64 which is not the architecture being linked (i386): /Users/admin/Sumit_Projects/Iphone/SamartRecovery/Smart iPhoneRecovery ProAmit/libimobiledevice.4.dylib
Undefined symbols for architecture i386:
  "_CopyAppFile", referenced from:
      -[AppWindowController sendWhatsApp] in AppWindowController.o
      -[AppWindowController sendSMS] in AppWindowController.o
      -[AppWindowController sendCalls] in AppWindowController.o
      -[AppWindowController sendContacts] in AppWindowController.o
      -[AppWindowController sendCalendar] in AppWindowController.o
      -[AppWindowController sendNotes] in AppWindowController.o
  "_CreateAppFolder", referenced from:
      -[AppWindowController sendWhatsApp] in AppWindowController.o
      -[AppWindowController sendSMS] in AppWindowController.o
      -[AppWindowController sendCalls] in AppWindowController.o
      -[AppWindowController sendContacts] in AppWindowController.o
      -[AppWindowController sendCalendar] in AppWindowController.o
      -[AppWindowController sendNotes] in AppWindowController.o
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Please provide me solution how to compile this for 32 architecture. 

Regards

aburgh, 2015-01-16:
When building libimobiledevice, you need to pass `-arch i386` to clang for it to build for 32-bit. Set CFLAGS either in your environment before running `configure` or as an argument: `./configure CFLAGS="-arch i386"`.

girishk55, 2015-01-20:
Hi aburgh, 

Thanks for your response, 

I tried this command but it cannot able to compile and install library in my machine. I Used 
./configure CFLAGS="-arch i386"

but got these error. 
ld: warning: ignoring file /usr/local/lib/libusbmuxd.dylib, file was built for x86_64 which is not the architecture being linked (i386): /usr/local/lib/libusbmuxd.dylib
ld: warning: ignoring file /usr/local/lib/libplist.dylib, file was built for x86_64 which is not the architecture being linked (i386): /usr/local/lib/libplist.dylib
ld: warning: ignoring file /usr/local/lib/libxml2.dylib, file was built for x86_64 which is not the architecture being linked (i386): /usr/local/lib/libxml2.dylib

Undefined symbols for architecture i386:
  "_afc_client_free", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_afc_client_new_with_service_client", referenced from:
      _afc_client_new_from_house_arrest_client in house_arrest.o
  "_afc_file_close", referenced from:
      _CopyAppFile in house_arrest.o
  "_afc_file_open", referenced from:
      _CopyAppFile in house_arrest.o
  "_afc_file_write", referenced from:
      _CopyAppFile in house_arrest.o
  "_afc_make_directory", referenced from:
      _CreateAppFolder in house_arrest.o
  "_afc_read_directory", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_idevice_free", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_idevice_new", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_idevice_set_debug_level", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_lockdownd_client_free", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_lockdownd_client_new_with_handshake", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_lockdownd_service_descriptor_free", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_lockdownd_start_service", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_plist_dict_get_item", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_plist_dict_set_item", referenced from:
      _house_arrest_send_command in house_arrest.o
  "_plist_free", referenced from:
      _house_arrest_send_command in house_arrest.o
      _house_arrest_get_result in house_arrest.o
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_plist_get_node_type", referenced from:
      _house_arrest_send_request in house_arrest.o
  "_plist_get_string_val", referenced from:
      _CopyAppFile in house_arrest.o
      _CreateAppFolder in house_arrest.o
  "_plist_new_dict", referenced from:
      _house_arrest_send_command in house_arrest.o
  "_plist_new_string", referenced from:
      _house_arrest_send_command in house_arrest.o
  "_property_list_service_client_free", referenced from:
      _house_arrest_client_free in house_arrest.o
  "_property_list_service_client_new", referenced from:
      _house_arrest_client_new in house_arrest.o
  "_property_list_service_receive_plist", referenced from:
      _house_arrest_get_result in house_arrest.o
  "_property_list_service_send_xml_plist", referenced from:
      _house_arrest_send_request in house_arrest.o
  "_service_client_factory_start_service", referenced from:
      _house_arrest_client_start_service in house_arrest.o
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libimobiledevice.la] Error 1
make: *** [install-recursive] Error 1

can you please help me that how i can compile the library without any error?

it is working on mac osx10.9 or higher version but it is not working with earlier version of the mac osx.

aburgh, 2015-01-20:
To compile libimobiledevice as 32-bit, all of its dependencies, i.e. the libraries it links against, must also be 32-bit. The three error messages like this are telling you that:

```
ld: warning: ignoring file /usr/local/lib/libusbmuxd.dylib, file was built for x86_64 which is not the architecture being linked (i386): /usr/local/lib/libusbmuxd.dylib
```

So, you need to first build `libplist`, `libusbmuxd`, and `libxml2` as 32-bit using the same CFLAGS tip as in my previous response.

Some other observations:

* Do you really need 32-bit? Mac OS X 10.6 (Snow Leopard) supports 64-bit on all Core 2 and newer Intel Macs, so the only ones that require 32-bit are the first Intel Macs, which used Intel Core Solo and Core Duo chips. Those are 8 or 9 years old now.

* Libxml2 is built into OS X, so you do not need to build it. However, there is no pkg-config `.pc` file, so the `configure` scripts don't find it. You can set environment variables to tell `configure` where the headers and libraries are located.

* If you do really need 32-bit, you will have to keep your 32-bit and 64-bit libraries separate or combine them into a fat binary. The `lipo` command is used to make fat binaries. Alternately, Apple's tools have support for automatically building fat binaries which works for some autotools  projects (I have no idea if the libimobiledevice projects will work). To get the automatic behavior, pass both the 32-bit and 64-bit flags to Clang, such as: `CFLAGS="-arch i386 -arch x86_64".

* The `file` tool will tell you if an executable or a library is 32-bit, 64-bit, or fat.

aburgh, 2015-01-20:
> can you please help me that how i can compile the library without any error?
> 
> it is working on mac osx10.9 or higher version but it is not working with earlier version of the mac osx.

Do you mean that the 64-bit tools you built will not run on versions of OS X earlier than 10.9? That's because you didn't set the minimum OS version when compiling. You can check like so:

```
$ otool -l /usr/local/bin/idevice_id | grep -A 3 MIN_MACOSX 
      cmd LC_VERSION_MIN_MACOSX
  cmdsize 16
  version 10.6
      sdk 10.10
```
In this example, the minimum OS X version idevice_id will run on is "10.6". To set this, add `-mmacosx-version-min=10.6` to the CFLAGS. See `man clang` for more details.

Jirmi, 2015-01-20:
Hi aburgh,
I'm also helping to girishk55 with the 32bit compilation. We have managed to move slightly forward with your advice. Thank you very much for that. 
Now we are experiencing issue when we try to compile libplist from source code. Please see below:-

```
libplist-master user$ sudo make
Password:
make  all-recursive
Making all in libcnary
make[2]: Nothing to be done for `all'.
Making all in src
/bin/sh ../libtool --tag=CC   --mode=link gcc -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I/usr/local/include/libxml2  -arch i386 -L/usr/local/lib -lxml2  -version-info 3:0:0 -no-undefined   -o libplist.la -rpath /usr/local/lib base64.lo bytearray.lo hashtable.lo ptrarray.lo xplist.lo bplist.lo plist.lo ../libcnary/libcnary.la 
libtool: link: gcc -dynamiclib  -o .libs/libplist.3.dylib  .libs/base64.o .libs/bytearray.o .libs/hashtable.o .libs/ptrarray.o .libs/xplist.o .libs/bplist.o .libs/plist.o   -Wl,-force_load,../libcnary/.libs/libcnary.a  -L/usr/local/lib /usr/local/lib/libxml2.dylib -lpthread -lz -liconv -lm  -arch i386   -install_name  /usr/local/lib/libplist.3.dylib -compatibility_version 4 -current_version 4.0 -Wl,-single_module
ld: warning: ld: warning: ignoring file .libs/base64.o, file was built for unsupported file format ( 0xcf 0xfa 0xed 0xfe 0x 7 0x 0 0x 0 0x 1 0x 3 0x 0 0x 0 0x 0 0x 1 0x 0 0x 0 0x 0 ) which is not the architecture being linked (i386): .libs/base64.oignoring file .libs/bytearray.o, file was built for unsupported file format ( 0xcf 0xfa 0xed 0xfe 0x 7 0x 0 0x 0 0x 1 0x 3 0x 0 0x 0 0x 0 0x 1 0x 0 0x 0 0x 0 ) which is not the architecture being linked (i386): .libs/bytearray.o

ld: warning: ld: warning: ignoring file .libs/hashtable.o, file was built for unsupported file format ( 0xcf 0xfa 0xed 0xfe 0x 7 0x 0 0x 0 0x 1 0x 3 0x 0 0x 0 0x 0 0x 1 0x 0 0x 0 0x 0 ) which is not the architecture being linked (i386): .libs/hashtable.oignoring file .libs/ptrarray.o, file was built for unsupported file format ( 0xcf 0xfa 0xed 0xfe 0x 7 0x 0 0x 0 0x 1 0x 3 0x 0 0x 0 0x 0 0x 1 0x 0 0x 0 0x 0 ) which is not the architecture being linked (i386): .libs/ptrarray.o

ld: warning: ld: warning: ignoring file .libs/plist.o, file was built for unsupported file format ( 0xcf 0xfa 0xed 0xfe 0x 7 0x 0 0x 0 0x 1 0x 3 0x 0 0x 0 0x 0 0x 1 0x 0 0x 0 0x 0 ) which is not the architecture being linked (i386): .libs/plist.oignoring file ../libcnary/.libs/libcnary.a, file was built for archive which is not the architecture being linked (i386): ../libcnary/.libs/libcnary.a

Undefined symbols for architecture i386:
  "_base64decode", referenced from:
      _xml_to_node in xplist.o
  "_base64encode", referenced from:
      _node_to_xml in xplist.o
  "_byte_array_append", referenced from:
      _write_int in bplist.o
      _write_uint in bplist.o
      _write_real in bplist.o
.....
  "_ptr_array_add", referenced from:
      _serialize_plist in bplist.o
  "_ptr_array_free", referenced from:
      _plist_to_bin in bplist.o
  "_ptr_array_index", referenced from:
      _plist_to_bin in bplist.o
  "_ptr_array_new", referenced from:
      _plist_to_bin in bplist.o
ld: symbol(s) not found for architecture i386
collect2: ld returned 1 exit status
make[2]: *** [libplist.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

Seems to me like the libcnary is not compiled for i386 despite the fact we have used command:-
```
./configure CFLAGS="-arch i386"
```
Could you please help us with this new issue?

FunkyM, 2015-01-20:
This looks like a build issue to me and not an issue in libimobiledevice.
You should read up on compiling universal binaries with multiple architectures by setting the CFLAGS as @aburgh described.
@Jirmi Please open up a ticket in proper project "libplist".

closed 2015-01-20 by FunkyM

aburgh, 2015-01-20:
I'm responding on this ticket because it will be helpful for building libimobiledevice, too. This worked for me:

```
$ git clone https://github.com/libimobiledevice/libplist.git
$ cd libplist
$  env NOCONFIGURE=yes ./autogen.sh
$ mkdir ../libplist-build ; cd ../libplist-build
$ ../libplist/configure CFLAGS="-arch i386 -arch x86_64" CXXFLAGS="-arch i386 -arch x86_64" libxml2_CFLAGS=-I/usr/include/libxml2 libxml2_LIBS="-L/usr/lib -lxml2"
$ make V=1
$ file (find . -name "*.a")
./libcnary/.libs/libcnary.a: Mach-O universal binary with 2 architectures
./libcnary/.libs/libcnary.a (for architecture i386):	current ar archive random library
./libcnary/.libs/libcnary.a (for architecture x86_64):	current ar archive random library
./src/.libs/libplist++.a:    Mach-O universal binary with 2 architectures
./src/.libs/libplist++.a (for architecture i386):	current ar archive random library
./src/.libs/libplist++.a (for architecture x86_64):	current ar archive random library
./src/.libs/libplist.a:      Mach-O universal binary with 2 architectures
./src/.libs/libplist.a (for architecture i386):	current ar archive random library
./src/.libs/libplist.a (for architecture x86_64):	current ar archive random library
```
Finding for "*.dylib" shows that the dylibs are also fat. Note that I forgot that libplist has C++ classes, so you have to set the architecture in CXXFLAGS, too.

It looks like you re-ran `configure` without cleaning your build folder, hence libcnary was skipped in your build log:

```
Making all in libcnary
make[2]: Nothing to be done for `all'.
```
If you configure `libusbmuxd` and `libimobiledevice` similarly, it appears they may work.

Jirmi, 2015-01-20:
Thanks a lot aburgh! Now I've managed to compile libimobiledevice with all dependencies for i386 architecture. The only problem which I have now is that I had to use :-

```
CFLAGS="-arch i386" CXXFLAGS="-arch i386"
```
instead of 
```
CFLAGS="-arch i386 -arch x86_64" CXXFLAGS="-arch i386 -arch x86_64"
```

The reason for this change was below mentioned error which I couldn't fix.

```
my-mac-mini:libplist-build2 user$ make
make  all-recursive
Making all in libcnary
/bin/sh ../libtool --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I.. -I../../libplist/libcnary    -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../../libplist/libcnary/include -arch i386 -arch x86_64 -MT node.lo -MD -MP -MF .deps/node.Tpo -c -o node.lo ../../libplist/libcnary/node.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I../../libplist/libcnary -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../../libplist/libcnary/include -arch i386 -arch x86_64 -MT node.lo -MD -MP -MF .deps/node.Tpo -c ../../libplist/libcnary/node.c  -fno-common -DPIC -o .libs/node.o
llvm-gcc-4.2: -E, -S, -save-temps and -M options are not allowed with multiple -arch flags
make[2]: *** [node.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

Any idea how to use multiple architecture's flags with using -M options?