libimobiledevice/libimobiledevice #172: IOS 8.1.2 libidevicemobile 1.1.9 ubuntu 14.04 LTS  lockdown , error code -2

mambo06, 2015-01-20:
compile goes very smooth and good.
when running program
use usbmoxd 1.1
always return lockdown , error code -2 for all application except for the "idevicepair pair", its says success but still no luck with other application, even idevicepair unpair return same error.
try to use usbmoxd 1.0.9 same error.
use -d option doesn't show anything

yangdm0209, 2015-02-02:
did you solve this problem?

mambo06, 2015-02-02:
Yea. By moving to arch. I guess its a 14.04 bug

yangdm0209, 2015-02-03:
Thank you very much.

Afforess, 2015-02-07:
I had the same issue on Ubuntu 14.10. I uninstalled the libimobiledevice package and instead built it from source. This resolved the issue, so it may be some kind of packaging issue.

Schroedingers-Cat, 2015-02-18:
Same problem. Compiled libimobiledevice and usbmuxd from source, but I get lockdown error code -2. On Ubuntu 14.04 and Ubuntu 14.10 however, you cannot remove the package libimobiledevice as it also removes the Ubuntu desktop or breaks dependencies if forced. @Afforess, how did you uninstall libimobiledevice?

Afforess, 2015-02-18:
I am not sure what you are talking about @Schroedingers-Cat, libimobiledevice is not a core package, nor part of ubuntu's base install.

sudo apt-get purge libimobiledevice* should remove it. When you run that, what packages does aptitude say you are uninstalling?

Schroedingers-Cat, 2015-02-18:
I have two Ubuntu installations where it seems to be the case.

On Ubuntu Gnome 14.10:
```sh
sudo apt-get purge libimobiledevice4
The following packages will be REMOVED:
  deja-dup-backend-gvfs* gdm* gnome-applets* gnome-power-manager*
  gnome-session* gnome-session-bin* gnome-session-flashback* gnome-shell*
  gnome-shell-extensions* gvfs-backends* gvfs-backends-goa* indicator-power*
  libgpod-common* libgpod4* libimobiledevice4* nautilus-share*
  rhythmbox-plugins* software-center* steadyflow* ubuntu-gnome-desktop*
  unity-control-center* unity-control-center-signon* upower* usbmuxd*
```

On Ubuntu 14.04:
```sh
sudo apt-get purge libimobiledevice4
The following packages will be REMOVED:
  deja-dup-backend-gvfs* gdm* gnome-applets* gnome-power-manager*
  gnome-session* gnome-session-bin* gnome-session-flashback* gnome-shell*
  gnome-shell-extensions* gvfs-backends* gvfs-backends-goa* indicator-power*
  libgpod-common* libgpod4* libimobiledevice4* nautilus-share*
  rhythmbox-plugins* software-center* steadyflow* ubuntu-gnome-desktop*
  unity-control-center* unity-control-center-signon* upower* usbmuxd*
```

Afforess, 2015-02-19:
Hm, that is quite bizarre. I suppose you could manually build the project, then try to find where the existing binaries are, and symlink those to your build output.

Schroedingers-Cat, 2015-02-19:
I just checked what packages will be removed if you purge libimobiledevice4 from Ubuntu 14.04/Ubuntu Gnome 14.10 Live CD: the same as I wrote above. So it's Ubuntus default behaviour. That leads to the question, if there even is an "official" way to compile and run libimobiledevice from source under Ubuntu.

Afforess, 2015-02-19:
@Schroedingers-Cat 

What you are seeing is very strange. I swear I had no issues uninstalling the package from Ubuntu 14.10. I will set up a VM later and see if I can reproduce the issue you see when you try to purge it.

stevesmename, 2015-02-22:
debugging found that usbmuxd_read_pair_record is returning -1 and flagging USERPREF_E_INVALID_CONF (error code -2). Believe the solution will be to update usbmuxd, the latest version from Ubuntu looks to be 1.0.8-2ubuntu1 but this version is three years out of date.