libimobiledevice/libimobiledevice #177: idevicediagnostics shutdown

dnsBlah, 2015-02-05:
When giving the shutdown command to an iPhone 6 or 6+ the shutdown command.
It does shutdown, but the unit turns back on when you pull the cable out.

Is this common with iPhone 6? It doesnt seem an libimobiledevice issue, just wondering.

aburgh, 2015-02-05:
I see that behavior, too, with an iPhone 6+: connecting or disconnecting from USB triggers the device to power on. Devices have powered on when connected to USB since the original iPhone, but this seems to be new. I have an iPhone 5 here, for example, and it did not power on when disconnected.

dnsBlah, 2015-12-08:
Sorry for late response again.
This is expected behavior of the unit. When the device is fully shutdown, it will be triggered to turn on after you unplug it.

Maybe Apple liked the idea that if you turned of your device while charging, that you would like to have it on when you unplug it :P

closed 2015-12-08 by dnsBlah