libimobiledevice/libimobiledevice #180: mobilebackup2 denied working after receiving of the first file

AlexxxRu, 2015-02-17:
Good afternoon!
I've tried to use mobilebackup2 but I have received some errors during downloading a file from the device.
 I'm using libimobiledevice for Windows. My operating system is Windows 8.1. 
Could you please help me in solving this problem?

**Little log of received headers:**
>Service starts on 50594 port
>Remote version: 2.1
>DLMessageDownloadFiles 	       //Request udid/Status.plist
>DLContentsOfDirectory  	       //Content of udid/Snapshot
>DLContentsOfDirectory  	       //Content of udid
>DLMessageGetFreeDiskSpace    //Free space 
>DLMessageCreateDirectory         //Create udid 
>DLMessageCreateDirectory	       //Create udid/Snapshot
>DLMessageUploadFiles               //Upload files

Then receiving device filename:
Smth like this: /var/tmp/tmp.2.5YnM5D
And correct name for my pc:
udid/Status.plist

Finally receiving  binary data and write to file. 
After that i call mobilebackup2_receive_message:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/
PropertyList-1.0.dtd">
<plist version="1.0">
<array>
        <string>DLMessageUploadFiles</string>
        <dict>
                <key>DLDeviceIOCallbacks</key>
                <integer>4955806704</integer>
        </dict>
        <real>13.061936</real>
        <integer>6878833</integer>
</array>
</plist>
```
After that I couldn't call anything. 
mobilebackup2_receive_raw all time return -1;
mobilebackup2_receive_message all time return -3;

Below are functions pseudocode and syslog from iphone.
Syslog:
```
Feb 17 03:12:14 iPhone BackupAgent2[1143] <Warning>: WARNING: Starting backup
Feb 17 03:12:14 iPhone BackupAgent2[1143] <Warning>: INFO: DeviceID="0b95d0db0c00fc9c5ab71d9f97299c35b3dca149", DeviceName="iPhone", ProductType="iPhone7,2", BuildVersion="12B411"
Feb 17 03:12:14 iPhone BackupAgent2[1143] <Notice>: (Note ) MC: User has completed cloud configuration. Not showing UI again.
Feb 17 03:12:14 iPhone installd[44] <Notice>: 0x100484000 -[MIClientConnection listSafeHarborsOfType:withOptions:completion:]: List safe harbor requested by client BackupAgent2 (pid 1143) for type 2 with options (null)
Feb 17 03:12:14 iPhone BackupAgent2[1143] <Warning>: INFO: Backup status: <MBStatus: version=2.4, date=2015-02-17 10:43:11 +0000, backupState=empty, snapshotState=uploading, fullBackup=1>
Feb 17 03:12:14 iPhone BackupAgent2[1143] <Warning>: INFO: Resuming after failure during 'uploading' phase
Feb 17 03:12:16 iPhone BackupAgent2[1143] <Warning>: INFO: Scanned in 1.500 s
Feb 17 03:12:16 iPhone BackupAgent2[1143] <Error>: 1143:29786112|SocketStreamHandler.c:_SocketLogCallback| ERROR: SocketSend (data): error 32: Broken pipe
Feb 17 03:12:16 iPhone BackupAgent2[1143] <Error>: 1143:29786112|StreamHandler.c:_WriteFunc| ERROR: Error sending data: Can't send data
Feb 17 03:12:16 iPhone BackupAgent2[1143] <Error>: 1143:29786112|DeviceLinkConnection.c:_DLSendBytesAndDisconnectOnError| ERROR: Could not send data: 'Error writing length of data'
Feb 17 03:12:16 iPhone BackupAgent2[1143] <Error>: 1143:29786112|DeviceLinkConnection.c:_DLSendBytesAndDisconnectOnError| ERROR: Could not send data: 'StreamHandler is NULL'
Feb 17 03:12:16 iPhone BackupAgent2[1143] <Warning>: ERROR: Computer disconnected: Lost connection
```
**Receiving file name:**
```
uint64_t fname_size = 0;
error = mobilebackup2_receive_raw(mobilebackup, (char*)&fname_size, 4, &bytes);
/*Error handling*/
/*Allocate memory for name*/
fname_size = be32toh(fname_size);
error = mobilebackup2_receive_raw(mobilebackup, *name, fname_size, &bytes);
/*Error handling*/
```
**Receiving file binary:**
```
uint32_t code_len = 0;
char *binary_data = NULL;
char *code = NULL;
error = mobilebackup2_receive_raw(mobilebackup, (char*)&code_len, 4, &bytes);
/*Error handling*/
code_len = be32toh(code_len);
/*Allocate memory for binaty_data*/

error = mobilebackup2_receive_raw(mobilebackup, &code, 1, &bytes);
/*Recv file code. First it '0x0c'*/
error = mobilebackup2_receive_raw(mobilebackup, binary_data, code_len-1, &bytes);
/*Error handling*/
/*Write to file*/
error = mobilebackup2_receive_raw(mobilebackup, &code, 1, &bytes);
/*Recv file code. Here '0x0'*/
/*raw_recive left data, 8 bytes 0x00*/
/*send response with empty dict and 0 status*/
```

nikias, 2015-02-17:
Did you take a look at idevicebackup2 code? It Shows you how it works.

AlexxxRu, 2015-07-13:
Error was in receiving files. Fixed by replace with mb2_handle_receive_files from idevicebackup2 code.

closed 2015-07-13 by AlexxxRu

FunkyM, 2015-07-13:
Great you worked it out. Please mind the LGPL license of the code if you copy it.
I think it might make sense to even refactor some of the file code into the mb2 service directly as helpers.