libimobiledevice/libimobiledevice #183: error while loading shared libraries: libimobiledevice.so.6

rageworx, 2015-03-02:
Dear all.
I've download latest tar.bz2 and build it in noticed sequence.
Matched required libraries, then succeed to build all and install it in root account by using sudo.
Checked is it installed well in /usr/local/lib and .bashrc - $PATH also applied well.
libimobiledivice.so and ~.so.6 was existed in symlink from libimobiledevice.so.6.0.0.
But ifuse or its tools are can't find libimobiledevice.so.6 like this:
<pre>
ifuse: error while loading shared libraries: libimobiledevice.so.6: cannot open shared object file: No such file or directory
</pre>
Is there any missing what I didn't, or something error ?
Thanks :)

nikias, 2015-03-15:
Libraries on Linux are not loaded from $PATH. Instead the loader is checking in configured paths depending on the ld config (usually /etc/ld.so.conf).
Often /usr/local/lib is already configured, but the library cache has not been updated. Try the command ```ldconfig```. Closing since this is not an issue with libimobiledevice itself.

closed 2015-03-15 by nikias

sandy0096, 2020-09-27:
sudo apt-get install \
	build-essential \
	checkinstall \
	git \
	autoconf \
	automake \
	libtool-bin \
	libplist-dev \
	libusbmuxd-dev \
	libssl-dev \
	usbmuxd

snowjobgit, 2021-06-23:
`sudo ldconfig` after all installations helped me