libimobiledevice/libimobiledevice #187: idevicebackup2 ErrorCode 212 Unable to backup until Setup is finished

mjkokid, 2015-03-30:
This error usally happen with iOS 8.1.2, 8.2. Using idevicebackup2 to backup iPhone 5 (8.2), it was failed, repeat 3-4 time it backuped success.
  libimobiledevice 1.1.7
  libplist 1.12
  libusbmuxd 1.0.10

mjkokid, 2015-03-30:
Started "com.apple.mobilebackup2" service on port 49225.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Backup will be unencrypted.
Requesting backup from device...
Incremental backup mode.
Backup Start..................
ErrorCode 212: Unable to backup until Setup is finished (MBErrorDomain/212)
Received 0 files from device.
Backup Failed (Error Code 212).

Louti, 2015-04-14:
Hello,

This happens when your device is still not setup or stuck in the setup screen (where you select the language, activation, passcode, ....) 

Cheers

nikias, 2015-05-23:
@Louti 100% correct. Closing.

closed 2015-05-23 by nikias