libimobiledevice/libimobiledevice #188: IOS 8.0 + Pi ERROR: Could not connect to lockdownd, error code -21

silencezt, 2015-04-02:
xxx@raspberry-pi:~# ideviceinfo
ERROR: Could not connect to lockdownd, error code -21
please help me ! Thanks!

FunkyM, 2015-04-08:
Depending on the version of libimobiledevice used this is either LOCKDOWN_E_USER_DENIED_PAIRING (older version) or LOCKDOWN_E_INVALID_HOST_ID (latest git master).
Latter means that your host id is screwed up, verify /var/lib/lockdown/SystemConfiguration.plist and remove it if necessary.
The other error code would indicate that you have denied the trust with this host in the trust dialog on the device before. If that is the case "idevicepair unpair" could help.
In any case, if creating tickets, please use descriptive titles and at least provide some details about your platform and library versions in use.

heisenfix, 2015-04-13:
Linux moe 3.18.3-201.fc21.x86_64 #1 SMP Mon Jan 19 15:59:31 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

ActivationState: Unactivated
BasebandActivationTicketVersion: V2
BasebandCertId: 3840149528
BasebandChipID: 8343777
BasebandKeyHashInformation: 
 AKeyStatus: 2
 SKeyHash: u+/tcCwvaQ+1Y9t40I4yegCEmB28mALlaROhaIVGBWo=
 SKeyStatus: 0
BasebandMasterKeyHash: 8CB15EE4C8002199070D9500BB8FB183B02713A5CA2A6B92DB5E75CE15536182
BasebandRegionSKU: BQAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
BasebandSerialNumber: CVyiBA==
BasebandStatus: BBInfoAvailable
BasebandVersion: 1.04.00
BluetoothAddress: 18:f6:43:3e:17:ff
BoardId: 6
BrickState: true
BuildVersion: 12D508
CPUArchitecture: arm64
CarrierBundleInfoArray[1]: 
 0: 
  CFBundleIdentifier: com.apple.Maroc_ma
  CFBundleVersion: 18.0
  IntegratedCircuitCardIdentity: 8921201013209356772
  InternationalMobileSubscriberIdentity: 604011320935677
  MCC: 604
  MNC: 01
  SIMGID1: /////w==
  SIMGID2: /////w==
CertID: 3840149528
ChipID: 28672
ChipSerialNo: CVyiBA==
CompassCalibration: AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
DeviceClass: iPhone
DeviceColor: #e1e4e3
DeviceName: iPhone
DevicePublicKey: LS0tLS1CRUdJTiBSU0EgUFVCTElDIEtFWS0tLS0tCk1JR0pBb0dCQU1GcWNUQStQb3o4WWpQSEFTcllTcFpaT09pamQ1S0ZRYi9RUmRvN1UrL1duMUNMZmYvS0hvQjQKYW1DZ2tWUVExRXZpZ0FXTzEzb2gyMDE3b1BPa3d5T081eXdzbDE4enovSmVjL1NEbjI5Rnl4TUl4dGdRSmtxawoxNS9NNkM3aWdwVzVEU0xVbEJyLytOa1A1cW8rNFRFRE11V0MwSVlDL0xOYWN0VHZqd2pOQWdNQkFBRT0KLS0tLS1FTkQgUlNBIFBVQkxJQyBLRVktLS0tLQo=
DieID: 3968258402871334
EthernetAddress: 18:f6:43:3e:18:00
FirmwareVersion: iBoot-2261.5.64
FirstFreePairExpired: true
FusingStatus: 3
HardwareModel: N61AP
HardwarePlatform: t7000
HostAttached: true
IntegratedCircuitCardIdentity: 8921201013209356772
InternationalMobileEquipmentIdentity: 356960066737243
InternationalMobileSubscriberIdentity: 604011320935677
MLBSerialNumber: C7H45050102F98FA6
MobileEquipmentIdentifier: 35696006673724
MobileSubscriberCountryCode: 604
MobileSubscriberNetworkCode: 01
ModelNumber: MG492
NonVolatileRAM: 
 auto-boot: dHJ1ZQ==
 backlight-level: MTUzMA==
 boot-args: 
 bootdelay: MA==
PartitionType: 
PasswordProtected: false
PkHash: 5OQIGNymupBn16zMKPujMp3562XDnNFkULy+gshbERM=
ProductType: iPhone7,2
ProductVersion: 8.2
ProductionSOC: true
ProtocolVersion: 2
ProximitySensorCalibration: T00EAA0LPzgQA7wCsASPAR8D7QADAKaY2AMDAKgT6wLuAkMAXm+KAgIEQhoAAAAA4QCVAAQCAAB8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
RegionInfo: ZD/A
SBLockdownEverRegisteredKey: false
SIMGID1: /w==
SIMGID2: /w==
SIMStatus: kCTSIMSupportSIMStatusReady
SIMTrayStatus: kCTSIMSupportSIMTrayInsertedWithSIM
SerialNumber: C7JNW3F4G5MQ
SoftwareBehavior: BQQAAAAAAAAAAAAAAAAAAA==
SoftwareBundleVersion: 
SupportedDeviceFamilies[1]: 
 0: 1
TelephonyCapability: true
TimeIntervalSince1970: 1428960990.754381
TimeZone: Africa/Casablanca
TimeZoneOffsetFromUTC: 3600.000000
TrustedHostAttached: true
UniqueChipID: 3968258402871334
UniqueDeviceID: 9d568a35bae33d0e2dd63d7b1c537f80f7d17188
UseRaptorCerts: true
Uses24HourClock: false
WiFiAddress: 18:f6:43:3e:17:fe
WirelessBoardSerialNumber: 4446A382068
kCTPostponementInfoPRIVersion: 0.1.141
kCTPostponementInfoPRLName: 0
kCTPostponementStatus: kCTPostponementStatusReady

 ideviceactivation activate
idevice_activation_request_new_from_lockdownd: Unable to get ActivationInfo from lockdownd
Failed to create activation request.

Any help FunkyM.

Thanks in advance.

heisenfix, 2015-04-13:
Apr 13 22:42:35 iPhone keybagd[39] <Error>: 0x100594000 handle_message: Command at index 12 fails as it needs proper entitlement
Apr 13 22:42:35 iPhone lockdownd[67] <Error>: call_and_response: Entitlement check failed in keybagd for  program(67): lockdownd

nikias, 2015-05-23:
@silencezt my guess is that the machine is too slow when generating the keypairs. This is what I observed on my Pi. Can you try if it works with this preliminary fix: https://dl.dropboxusercontent.com/u/13629465/limd-slow-keygen-fix.diff ?

jasonsilver, 2015-07-15:
I'm having the same problem, but my error code is -2. I'm using an iphone 4 with iOS7 and Ubuntu 14.04 LTS. I compiled all of the components myself.

My command:
- idevicepair unpair
- ERROR: Device 7ec15b69bf887499a6a92ab905bb95088c694a50 returned unhandled error code -2

Interesting, so I pair it and get:
- ERROR: Please accept the trust dialog on the screen of device [HASH], then attempt to pair again.

I do it again and get:
- SUCCESS: Paired with device 

Then about 1.5 seconds later the iphone dings and seems to lose the pair... that's my assumption, since if I try to unpair it I get the same error above (-2), or if I pair it again, I have to accept the trust dialogue all over again.

Running ios_webkit_debug_proxy gives me the following messages:
 - Listing devices on :9221
 - Could not connect to lockdownd. Exiting.: Operation now in progress
 - Unable to attach 7ec15b69bf887499a6a92ab905bb95088c694a50 inspector

I've tried running this quickly after pairing (before the iphone dings) but it doesn't seem to affect the outcome.

I've found numerous suggestions online and have tried everything that seemed reasonable, but so far no luck.

BrandonHe, 2016-11-22:
@FunkyM @nikias 
I have same problem with "error code -21", according to your suggestions, I use latest git master and have tried to remove SystemConfigure.plist, also, I am sure /var/lib/lockdown/ has both right and read access, but unluckily, it's still not work.
> Depending on the version of libimobiledevice used this is either LOCKDOWN_E_USER_DENIED_PAIRING (older version) or LOCKDOWN_E_INVALID_HOST_ID (latest git master).
Latter means that your host id is screwed up, verify /var/lib/lockdown/SystemConfiguration.plist and remove it if necessary

I have selected trust in the trust dialog, here are more details attached:
```
$ idevicepair unpair
ERROR: Device 94062c249078ef7931bf08ee83a2ba730bc7e352 is not paired with this host
$ idevicepair pair
SUCCESS: Paired with device 94062c249078ef7931bf08ee83a2ba730bc7e352
$ ideviceinfo 
ERROR: Could not connect to lockdownd, error code -17
$ ideviceinfo -s
BasebandCertId: 3840149528
BasebandKeyHashInformation: 
 AKeyStatus: 2
 SKeyHash: u+/tcCwvaQ+1Y9t40I4yegCEmB28mALlaROhaIVGBWo=
 SKeyStatus: 0
BasebandSerialNumber: Cn4jog==
BasebandVersion: 5.24.00
BoardId: 6
BuildVersion: 14A456
ChipID: 28672
DeviceClass: iPhone
DeviceColor: #e1e4e3
DeviceName: iPhone
DieID: 8554561647182886
HardwareModel: N61AP
PartitionType: 
ProductName: iPhone OS
ProductType: iPhone7,2
ProductVersion: 10.0.2
ProductionSOC: true
ProtocolVersion: 2
TelephonyCapability: true
UniqueChipID: 8554561647182886
UniqueDeviceID: 94062c249078ef7931bf08ee83a2ba730bc7e352
WiFiAddress: 28:5a:eb:25:0d:d8
$ ifuse /media/iphone/
Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.
```

Tested with iOS 10.0.2 + Ubuntu 14.04 LTS
Any suggestion here? 

Thanks in advance!

BrandonHe, 2016-11-23:
@FunkyM @nikias 
Tried to install libimobiledevice on a new virtual machine Ubuntu 14.04-32, same issues with above post.
Add more infos:

| package| version |
|--------|-------|
| libplist | 1.13.0 |
| libusbmuxd | 1.1.0|
| libimobiledevice | 1.2.1 |
| usbmuxd | 1.1.1 |
| ifuse | 1.1.3 |
| ideviceinstaller |  |
| libmobiledevice | 1.2.1 |

Could you guys offer any help?

lionsguard, 2016-12-09:
I tried the fix posted by @nikias for the threading solution but I still get the "failed to read /var/lib/lockdown/..." message and the error dialog that pops up after plugging in my device and accepting the "trust" dialog varies between (-20) and (-256) and sometimes (-21), (-5), and (-3).

This in on a PI, v3. The way the Pi is being used is that iPads will be plugged in and removed with no user access to the Pi via GUI or command line. The Pi is an in-between device that users do not interact with other than having things plugged into it.

I compiled all the sources from HEAD in order to use OpenSSL and applied @nikias fix to lockdown.c but am still having these issues.

Any suggestions?

yangbiao01, 2017-01-10:
I have this problem in ubuntu14.04,but Solvedd in ubuntu 16.04.1

AlexJinsin, 2017-03-20:
I'm  mac os  10.12.3 . How can i fix this?

timwr, 2017-05-13:
If you've arrived here from google I fixed this by running with sudo, e.g `sudo ideviceinfo`. I installed libimobiledevice from brew.

tristan-k, 2018-07-08:
@timwr This doesnt work for me. I also tried `brew install ideviceinstaller --HEAD` without any success. Anybody?

timwr, 2018-07-08:
@tristan-k try `sudo idevicepair pair`, or become root `sudo su`, then try again.