libimobiledevice/libimobiledevice #189: help!!!

silencezt, 2015-04-02:
xxx@raspberry-pi:~# ideviceinfo
ERROR: Could not connect to lockdownd, error code -21
please help me ! Thanks!

FunkyM, 2015-04-08:
Depending on the version of libimobiledevice used this is either LOCKDOWN_E_USER_DENIED_PAIRING (older version) or LOCKDOWN_E_INVALID_HOST_ID (latest git master).
Latter means that your host id is screwed up, verify /var/lib/lockdown/SystemConfiguration.plist and remove it if necessary.
The other error code would indicate that you have denied the trust with this host in the trust dialog on the device before. If that is the case "idevicepair unpair" could help.
In any case, if suppling tickets, please use descriptive titles and at least provide some details about your platform and library versions in use.

FunkyM, 2015-04-08:
Closing as it's a duplicate of #188.

closed 2015-04-08 by FunkyM