libimobiledevice/libimobiledevice #190: How can i get the usb port/path/address and udid in windows?

bestplay, 2015-04-09:
Hi, friends. thanks for your great job. 

I need got the usb path of each udid.

I can not match each udid with its usb path.(consider multiple devices)

using idevice_id.exe -l only get the udid list. 
and use libusb to get the usb path and maybe serialNumber.
[libusb_get_string_descriptor_ascii(dev, desc.iSerialNumber, (unsigned char*)strDesc, 128)]

but It seems the serialNumber is not an udid ? 

thanks,
Wayne.

jensolsson, 2018-04-07:
I am also looking for the same thing but on linux. Did you find  a solution?

mexmer, 2018-04-08:
why should udid be serial number?
serial number i unique for product model, while UDID is unique across all apple devices.

nikias, 2018-04-08:
The USB device serial number is the UDID. Not to be confused with Apple's device serial number. Two different things.

jensolsson, 2018-04-09:
@niklas so you mean using the UDID it is possible to find the usb path using libusb for example?

Is really the UDID a USB thing? I always thought it was for apple devices only and nothing to do with usb?

nikias, 2018-04-09:
No the UDID is not a USB thing. It's just Apple decided to use the iSerialNumber field for the UDID.

jensolsson, 2018-04-09:
@nikias are you 100% sure about this?
Using lsusb -v I get iSerial's that are all empty string for all Apple devices. so I agree with @bestplay that it does not seem like iSerial is the same as Apples UDID for iPhones. Or there is a bug somewhere in lsusb so it does not display the iSerial.

For all android devices the iSerial is the same number as I receive when running "adb devices" 

See for example below for one of my iPhones

#lsusb -v
<snip>
Bus 001 Device 088: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x05ac Apple, Inc.
  idProduct          0x12a8 iPhone5/5C/5S/6
  bcdDevice            9.03
  iManufacturer           1 
  iProduct                2 
  **iSerial                 3** 
  bNumConfigurations      4
  Configuration Descriptor:
<snip>

nikias, 2018-04-10:
You need to query iSerialNumber, not iSerial. See here how usbmuxd does it: https://github.com/libimobiledevice/usbmuxd/blob/master/src/usb.c#L339

nikias, 2018-04-10:
In case you are looking for a synchronous method, check the older code here:
https://github.com/libimobiledevice/usbmuxd/blob/b752fbc1dd92238a34c888fbaef921aa802c4a82/src/usb.c#L389

jensolsson, 2018-04-10:
@nikias I dont understand this. 

Looking at the source code for lsusb for example, dont know if this is the official repo, but looks like the righ code:
https://github.com/gregkh/usbutils/blob/master/lsusb.c

I see on line 285:
serial = get_dev_string(dev, descriptor->iSerialNumber);

Then I see on line 287:
printf("Device Descriptor:\n"
	       "  bLength             %5u\n"
	       "  bDescriptorType     %5u\n"
	       "  bcdUSB              %2x.%02x\n"
	       "  bDeviceClass        %5u %s\n"
	       "  bDeviceSubClass     %5u %s\n"
	       "  bDeviceProtocol     %5u %s\n"
	       "  bMaxPacketSize0     %5u\n"
	       "  idVendor           0x%04x %s\n"
	       "  idProduct          0x%04x %s\n"
	       "  bcdDevice           %2x.%02x\n"
	       "  iManufacturer       %5u %s\n"
	       "  iProduct            %5u %s\n"
	       **"  iSerial             %5u %s\n"**
	       "  bNumConfigurations  %5u\n",
	       descriptor->bLength, descriptor->bDescriptorType,
	       descriptor->bcdUSB >> 8, descriptor->bcdUSB & 0xff,
	       descriptor->bDeviceClass, cls,
	       descriptor->bDeviceSubClass, subcls,
	       descriptor->bDeviceProtocol, proto,
	       descriptor->bMaxPacketSize0,
	       descriptor->idVendor, vendor, descriptor->idProduct, product,
	       descriptor->bcdDevice >> 8, descriptor->bcdDevice & 0xff,
	       descriptor->iManufacturer, mfg,
	       descriptor->iProduct, prod,
	       **descriptor->iSerialNumber, serial,**
	       descriptor->bNumConfigurations);

So iSerial printed by lsusb is indeed iSerialNumber

But iSerial seem empty for iPhone devices as I wrote above.

mexmer, 2018-04-10:
@jensolsson 
i suspect there is problem on your side, i just checked on recent stable debian, and lsusb indeed shows UDID under iSerial. and i'm currently not even runing usbmuxd, i just booted it to check your claims.

as for windows, UDID is part of device instance path, although if you iterate trough all usbdevices, and filter only those who match APPLE VID, you can simply ask them for description string for serial number property, and it will return UDID.

still this is apple decision to supply UDID on serial number descriptor request.

jensolsson, 2018-04-10:
Oh, never thought there would be a problem on my machine.

Running a dmesg indeed shows the UDID number so I have no clue why lsusb would not.

Maybe I should just read dmesg backwards to find the usb path, but it does not feel optimal.

Here is my lsusb version. dont know if it is old or something. installed from packages.
jensolsson@host:~$ lsusb --version
lsusb (usbutils) 007

[2549902.396887] usb 1-1.4.3.3: SerialNumber: 6ca03dc3d04adf64572<hidden>

Here is my info:
jensolsson@host:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful

jensolsson@host:~$ lsusb
Bus 002 Device 041: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 002 Device 042: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 002 Device 030: ID 18d1:4ee2 Google Inc. Nexus 4 (debug)
Bus 002 Device 037: ID 18d1:4ee7 Google Inc. 
Bus 002 Device 028: ID 2109:2811 VIA Labs, Inc. Hub
Bus 002 Device 045: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 002 Device 046: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 002 Device 003: ID 2109:2811 VIA Labs, Inc. Hub
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 067: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 001 Device 091: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 001 Device 077: ID 18d1:4ee7 Google Inc. 
Bus 001 Device 080: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 001 Device 062: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 089: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 001 Device 004: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 070: ID 0fce:51d9 Sony Ericsson Mobile Communications AB 
Bus 001 Device 068: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 001 Device 087: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 001 Device 064: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 001 Device 052: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 088: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 001 Device 092: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy (MTP)
Bus 001 Device 003: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

mexmer, 2018-04-10:
Might be issue with privileges, if you run lsusb -v as root, do you still get “unable to open device” on iphone?

jensolsson, 2018-04-10:
oh, it was that simple? Works perfectly. Thanks for helping me out!