libimobiledevice/libimobiledevice #191: Can't pair with iphone correctly from virtual machine

zhangfan2012fall, 2015-04-09:
All libs are built from github master branch.
ideviceinfo failed with error code -21.
`/var/lib/lockdown` is empty

```user@user-VirtualBox:~$ idevicepair -d pair
SUCCESS: Paired with device [sha]
user@user-VirtualBox:~4 idevicepair validate
ERROR: Device [sha] is not paired with this host
user@user-VirtualBox:/usr$ idevice_id -l
[sha]
user@user-VirtualBox:/usr$ ideviceinfo
ERROR: Could not connect to lockdownd, error code -21```

nikias, 2015-05-23:
Hi! my guess is that the machine is too slow when generating the keypairs. Can you try if it works with this preliminary fix: https://dl.dropboxusercontent.com/u/13629465/limd-slow-keygen-fix.diff ?

nikias, 2018-12-16:
Updated the link.