libimobiledevice/libimobiledevice #193: Cannot create HouseArrestClient on VendDocuments

kekiefer, 2015-04-13:
This problem manifested itself on ifuse originally (https://github.com/libimobiledevice/ifuse/issues/24), but I have the same issue using HouseArrestClient myself.

```python
device = imobiledevice.iDevice(serial)
ha_lockdown = imobiledevice.LockdownClient(device, 'myapp').start_service(imobiledevice.HouseArrestClient)
ha_client = imobiledevice.HouseArrestClient(device, ha_lockdown)
ha_client.send_command('VendDocuments', 'com.mycompany.myapp')
result = ha_client.get_result()
print "Initialized imobiledevice HouseArrestClient: %s" % (result)
```

And what I get
```
Initialized imobiledevice HouseArrestClient: {'Error': u'InstallationLookupFailed'}
```

Using `VendContainer` works fine, so there's a workaround (for my use case at least). I'm not sure what this error means. Is it just that Apple aren't supporting `VendDocuments` anymore?

mingio, 2015-04-14:
Hi, kekiefer
I get the InstallationLookupFailed error too when I use VendDocuments. But I use VendContainer and get the same error. Are u sure the VendContainer can work on ios8.3?

kekiefer, 2015-04-14:
Yes, though this was with just one app on two iPads. One interesting note here is that @dabvid from the issue referenced above reported that VendDocuments worked for him and VendContainer did not. Maybe there is more to this.

kekiefer, 2015-04-14:
So a testflight build is working OK, but it turns out my workaround failed with the iTunes version of the app. The real situation may be rather dire:
http://www.iclarified.com/48418/ios-83-blocks-file-manager-software-from-accessing-app-data
http://www.macrumors.com/2015/04/13/ios-8-3-ifunbox-itools-sandbox-app-access/

closed 2015-04-15 by kekiefer

dunhamsteve, 2015-04-22:
VendContainer does appear to work for apps that are installed via either AdHoc or XCode, but not for apps installed via the App Store.  (Tested via ifuse --container.)

kekiefer, 2015-04-22:
Thank you @dunhamsteve -- this does appear to be the case and it makes a lot more sense now.

theonlylawislove, 2015-05-28:
Does anybody how to access the file system with the "UIFileSharingEnabled" method? HouseArrest has no knowledge of where I intend to modify the file system, so since that method is now broken, what  other method is there to access the "/Documents" folder?

mexmer, 2015-05-28:
in 8.3 they changed behavior of HA filesystem.
VendDocuments works only for applications that have FileSharingUI enabled
aditionaly you cannot list root folder of HA (it gives access denied error), but you can cd into /Documents folder, there you can work like before.

theonlylawislove, 2015-05-29:
Hey guys, I started making a stub 64bit iTunesMobileDevice.dll that forwards methods to the real iTunesMobileDevice.dll so that we can intercept and see how some closed source applications are still allowing us to write to the iOS 8.3 shared "Documents" folder.

https://github.com/theonlylawislove/iTunesMobileDevice-Hook

iFunBox is using "AMDeviceCreateHouseArrestService" instead of "AMDeviceStartHouseArrestService". Maybe the former gives more options, letting us to HouseArrest an app in "Documents" mode, instead of "Container" mode.

Has anyone had any experience with the AMDeviceCreateHouseArrestService method? How is iFunBox using it to access an iPad app's documents directory?

mexmer, 2015-05-29:
@theonlylawislove as i stated above you can write to /Documents in HA on 8.3, but you can start HA VendDocument only on applications that have filesharing enabled.
also as i said, you can't list content of /

you CD into /Documents, and work there ... that is ... works fine.