libimobiledevice/libimobiledevice #199: libimobiledevice mailing-list not transferring all emails

saintger, 2015-04-22:
Hello,

It seems that the current mailing-list libimobiledevice-devel@lists.libimobiledevice.org has a bug.
Some emails are not transferred to members, so nobody get the chance to answer.

For instance I haven't received the message from Adrian Stanley on 2015-04-17.
And it seems that nobody received my messages on 2015-03-07 and 2015-04-19 (confirmed by someone on the freenoce IRC).

Should I rather use Github to communicate ?

Thanks

cfergeau, 2015-04-22:
I was the one on IRC who did not get the 2015-04-17 emai (which is present in the archives). As we both are using GMail, this could be libimobiledevice.org having issues sending emails to GMail (blacklisted ?). I tried sending a password reminder to both my gmail address, and an address not managed by gmail. I've received the password reminder to my other address, but not to the GMail one.

saintger, 2015-04-22:
I've checked my spam folder and the message from 2015-04-17 is also not there.
Can you see my messages from 2015-03-07 and 2015-04-19 ?

cfergeau, 2015-04-22:
March 7th is in my libimobiledevice GMail folder, April 19th is not. The last email I got from the mailing list was on April 8th even though there are 10 to 20 emails in the archives from after that date.

saintger, 2015-04-22:
I got all except the one from April 8th.
And you have nothing in your spam folder ?
After a quick search on Mailman and Gmail relation, it seems to be a bouncing problem from Gmail side.
However with no access to the server, it is difficult for me to do anything except reporting the problem... :-(

cfergeau, 2015-04-22:
Ah right, starting from April, GMail decided to put libimobiledevice emails in the spam folder. I've got both your messages, and the iOS 8.3 one now.

FunkyM, 2015-04-23:
Yes, GMail decided to treat our mail as spam. Overall blacklist/RBL status for the server is fine but there must be something else that Google doesn't like. I'll probably investigate during the weekend.

saintger, 2015-04-25:
I also managed to find the missing email (2015-04-17) in the spam folder (I don't know why I haven't seen it before).

FunkyM, 2015-11-02:
Should be fixed now as the MTA setup has received love to be liked by GMail.

closed 2015-11-02 by FunkyM