libimobiledevice/libimobiledevice #200: configure: WARNING: cannot find libplist Cython bindings

kbulzo, 2015-04-23:
when trying to install libimobiledevice get this error checking consistency of all components of python development environment... yes
checking for libplist Cython bindings... no
configure: WARNING: cannot find libplist Cython bindings. You should install your distribution specific libplist Cython bindings package.
checking for openssl... no
configure: error: OpenSSL support explicitly requested but OpenSSL could not be found
O.S Ubuntu 14.04 LTS  how to fix?

rodrigodiaz, 2015-06-02:
Hello. The issue is openssl.
sudo apt-get install libssl-dev

ripper2hl, 2015-12-20:
@rodrigodiaz its working for me, why no close this issue ?

JosefJezek, 2018-12-17:
Yes, its working also on Ubuntu 18.04. Close it.

shawn7226791, 2019-11-04:
hey can someone help me when installing the samething it give me this

config.status: error: cannot find input file: `common/Makefile.in'