libimobiledevice/libimobiledevice #204: version1.2.0 can use mobilebackup on iOS8?

EvilLeo, 2015-04-30:
###  Version
libimobiledevice: 1.2.0
iOS: 8.1.2

when i use mobilebackup, **not mobilebackup2**, i can not use mobilebackup_client_new to get a client, 
the err is **MOBILEBACKUP_E_BAD_VERSION**

then i view the source code, it occuript here(mobilebackup.c->device_link_service.c):
``` 
 190     if (vmajor > version_major) {
 191         debug_info("Version mismatch: device=(%lld,%lld) > expected=(%lld,%lld)", vmajor, vminor, version_major, version_minor);
 192         err = DEVICE_LINK_SERVICE_E_BAD_VERSION;
 193         goto leave;
 194     } else if ((vmajor == version_major) && (vminor > version_minor)) {
 195         debug_info("WARNING: Version mismatch: device=(%lld,%lld) > expected=(%lld,%lld)", vmajor, vminor, version_major, version_minor);
 196         err = DEVICE_LINK_SERVICE_E_BAD_VERSION;
 197         goto leave;
 198     }
```

so it just can use mobilebackup2 on iOS8 ?
thx

aburgh, 2015-04-30:
Correct, `mobilebackup_*` is for iOS <= 3, `mobilebackup2_*` is for iOS >= 4.

nikias, 2015-05-23:
Closing. @aburgh answered the question.

closed 2015-05-23 by nikias