libimobiledevice/libimobiledevice #205: ERROR: Could not start service com.apple.mobilebackup2

briancdillard, 2015-04-30:
Running the libimobiledevice stack on Santoku-Linux in VMware. Connected to an iPad2 running iOS 8.1.3.

Everything works well up until I want to run the idevicebackup2 process.

I'm paired and I can even pull system info with the "ideviceinfo -s" command.

I have given all my directories full access, yet still no go.

Here's the return I get from every attempt:

$ idevicebackup backup -d ~/Documents/iPadBackups
Backup directory is "/home/Santoku/Documents/iPhoneBackups"
ERROR: Could not start service com.apple.mobilebackup2

Been working (and googling) this issue for days and can't find a solution. Any help is more than welcome and appreciated. Thanks.

aburgh, 2015-04-30:
Have you tried restarting the device? If that doesn't resolve it, start `idevicesyslog` in a second terminal session, then try the backup again and capture the log from `idevicesyslog`.

briancdillard, 2015-05-01:
Just ran the backup while running the syslog. Results:

May  1 11:37:55 tr10533s-iPad kernel[0] <Notice>: flow_divert_token_set (0): Failed to get the key unit from the token: 22
May  1 11:37:55 tr10533s-iPad lockdownd[25] <Error>: Could not set socket option SO_OPPORTUNISTIC: Invalid argument
May  1 11:37:55 tr10533s-iPad lockdownd[25] <Error>: Could not set socket option SO_OPPORTUNISTIC: Invalid argument
May  1 11:37:55 tr10533s-iPad kernel[0] <Notice>: flow_divert_token_set (0): Failed to get the key unit from the token: 22
May  1 11:37:55 tr10533s-iPad kernel[0] <Notice>: flow_divert_token_set (0): Failed to get the key unit from the token: 22
May  1 11:37:55 tr10533s-iPad lockdownd[25] <Error>: Could not set socket option SO_OPPORTUNISTIC: Invalid argument
May  1 11:37:55 tr10533s-iPad configd[24] <Notice>: DHCP en0: can't schedule wake-up, expiration is too soon
May  1 11:37:55 tr10533s-iPad configd[24] <Notice>: DHCP en0: now = 0x1af3e363, lease = { start 0x1af3e359, t1 0x1af3e363, t2 0x1af3e36a, expiration 0x1af3e36d }
May  1 11:37:55 tr10533s-iPad configd[24] <Notice>: DHCP en0: can't schedule wake-up, expiration is too soon
May  1 11:37:55 tr10533s-iPad configd[24] <Notice>: DHCP en0: now = 0x1af3e363, lease = { start 0x1af3e363, t1 0x1af3e36d, t2 0x1af3e374, expiration 0x1af3e377 }
May  1 11:37:55 tr10533s-iPad notification_proxy[145] <Error>: 0x31d000 -[MNPLockdownConnection receiveMessage]: lockdown_receive_message error!
May  1 11:38:00 tr10533s-iPad afcd[165] <Error>: Could not check in with lockdown at 'xpc': 1
May  1 11:38:00 tr10533s-iPad com.apple.xpc.launchd[1] (com.apple.afcd[165]) <Warning>: Service exited with abnormal code: 1
May  1 11:38:00 tr10533s-iPad BackupAgent2[166] <Error>: Could not check in with lockdown at 'xpc': 1

Going to spend some time looking over the results for a solution. Figured I would post it on the thread just in case anyone knows anything that will help reduce my time of troubleshooting. Thanks!

wputnam, 2015-06-17:
I am also having this issue. (in Fedora, with repository files)

    [user@localhost Desktop]$ idevicebackup2 backup --full ~
    Backup directory is "/home/user"
    ERROR: Could not start service com.apple.mobilebackup2.

I've tried restarting, pairing/unpairing, reinstalling the software, and running idevicesyslog and piping it to a text file while attempting to run the backup, but I haven't seen any relevant errors to this service.

In addition, I am unable to take a screenshot. I was able to earlier, but that changed once I tried changing my phone's lock settings to "Never" so that I could do the backup. 

    [user@localhost Desktop]$ idevicescreenshot
    Could not start screenshotr service! Remember that you have to mount the Developer disk image 
    on your device if you want to use the screenshotr service.

Other than that, everything works.

aburgh, 2015-06-18:
> In addition, I am unable to take a screenshot. I was able to earlier, but that changed once I tried changing my phone's lock settings to "Never" so that I could do the backup.

		[user@localhost Desktop]$ idevicescreenshot
		Could not start screenshotr service! Remember that you have to mount the Developer disk image 
		on your device if you want to use the screenshotr service.

The developer disk images are included in the Xcode bundle on OS X. `ideviceimagemounter` can be used to send it to the device and mount it. Or, if you connect the device to a machine with Xcode running, Xcode will do it automatically.

aburgh, 2015-06-18:
> I'm paired and I can even pull system info with the "ideviceinfo -s" command.

The `-s` option, "simple", skips the SSL handshake with `lockdownd` and thus works without pairing the device. If you try `ideviceinfo` without the `-s`, does it work?

wputnam, 2015-06-18:
Aaron, I know that the software is the problem because I have the same software installed on a separate machine (Ubuntu), and running a full backup on that machine gives me no problems whatsoever.

aburgh, 2015-06-18:
@wputnam I thought you were reporting the screenshot problem as a separate issue, but I see your point. It may or may not be related. Can you run `idevicebackup2` with the `-d` option and post the output?

wputnam, 2015-06-18:
@aburgh 

    [user@localhost ~]$ idevicebackup2 -d backup Desktop/
    Backup directory is "Desktop/"
    ERROR: Could not start service com.apple.mobilebackup2.
    [user@localhost ~]$ 

In addition, I reinstalled Fedora entirely yesterday evening, ran the repository updates, and installed the additional libimobiledevicetools. Nothing else. It is still doing this.

aburgh, 2015-06-24:
Can you recompile the library with debug info try the backup again? Pass `--enable-debug-code` to `./configure`.

Also, can you confirm that `idevice` works without the `-s` option?

wputnam, 2015-06-24:
@aburgh I was installing libimobiledevice through the repository with Fedora/openSUSE, so I don't think that I can handle the configure for that.

In the meantime, the ideviceinfo -s gives me plenty of information without any errors.

aburgh, 2015-06-27:
Sounds like a mismatch with dependencies, but I don't have a Fedora installion to test. My next step would be to try installing libimobiledevice and dependencies from the github sources.

wputnam, 2015-06-29:
@aburgh After a fresh install of Fedora, I installed the entire default development tools/libraries packages, plus libtool separately as it wasn't part of those packages. I then used the repository to install the libimobiledevicepackages and their dependencies:
    sudo dnf install libimobiledevice-*
With that done, I tried the backup again. No dice.

I then downloaded only the libimobiledevice package from GitHub and tried compiling it with that option:

    [user@localhost libimobiledevice-master]$ sudo ./autogen.sh --enable-debug-code
    libtoolize: putting auxiliary files in `.'.
    libtoolize: linking file `./ltmain.sh'
    libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
    libtoolize: linking file `m4/libtool.m4'
    libtoolize: linking file `m4/ltoptions.m4'
    libtoolize: linking file `m4/ltsugar.m4'
    libtoolize: linking file `m4/ltversion.m4'
    libtoolize: linking file `m4/lt~obsolete.m4'
    configure.ac:34: installing './config.guess'
    configure.ac:34: installing './config.sub'
    checking for a BSD-compatible install... /bin/install -c
    checking whether build environment is sane... yes
    checking for a thread-safe mkdir -p... /bin/mkdir -p
    checking for gawk... gawk
    checking whether make sets $(MAKE)... yes
    checking whether make supports nested variables... yes
    checking whether make supports nested variables... (cached) yes
    checking for gcc... gcc
    checking whether the C compiler works... yes
    checking for C compiler default output file name... a.out
    checking for suffix of executables... 
    checking whether we are cross compiling... no
    checking for suffix of object files... o
    checking whether we are using the GNU C compiler... yes
    checking whether gcc accepts -g... yes
    checking for gcc option to accept ISO C89... none needed
    checking whether gcc understands -c and -o together... yes
    checking for style of include used by make... GNU
    checking dependency style of gcc... gcc3
    checking for g++... no
    checking for c++... no
    checking for gpp... no
    checking for aCC... no
    checking for CC... no
    checking for cxx... no
    checking for cc++... no
    checking for cl.exe... no
    checking for FCC... no
    checking for KCC... no
    checking for RCC... no
    checking for xlC_r... no
    checking for xlC... no
    checking whether we are using the GNU C++ compiler... no
    checking whether g++ accepts -g... no
    checking dependency style of g++... none
    checking build system type... x86_64-unknown-linux-gnu
    checking host system type... x86_64-unknown-linux-gnu
    checking how to print strings... printf
    checking for a sed that does not truncate output... /bin/sed
    checking for grep that handles long lines and -e... /bin/grep
    checking for egrep... /bin/grep -E
    checking for fgrep... /bin/grep -F
    checking for ld used by gcc... /bin/ld
    checking if the linker (/bin/ld) is GNU ld... yes
    checking for BSD- or MS-compatible name lister (nm)... /bin/nm -B
    checking the name lister (/bin/nm -B) interface... BSD nm
    checking whether ln -s works... yes
    checking the maximum length of command line arguments... 1572864
    checking whether the shell understands some XSI constructs... yes
    checking whether the shell understands "+="... yes
    checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
    checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
    checking for /bin/ld option to reload object files... -r
[... 292 lines cut ...]
     /bin/install -c -m 644 idevice_id.1 ideviceinfo.1 idevicesyslog.1 idevicebackup.1 idevicebackup2.1 ideviceimagemounter.1 idevicescreenshot.1 idevicepair.1 ideviceenterrecovery.1 idevicedate.1 ideviceprovision.1 idevicedebugserverproxy.1 idevicediagnostics.1 idevicecrashreport.1 idevicename.1 idevicedebug.1 idevicenotificationproxy.1 '/usr/local/share/man/man1'
    make[2]: Leaving directory '/home/user/libimobiledevice-master/docs'
    make[1]: Leaving directory '/home/user/libimobiledevice-master/docs'
    make[1]: Entering directory '/home/user/libimobiledevice-master'
    make[2]: Entering directory '/home/user/libimobiledevice-master'
    make[2]: Nothing to be done for 'install-exec-am'.
    make[2]: Nothing to be done for 'install-data-am'.
    make[2]: Leaving directory '/home/user/libimobiledevice-master'
    make[1]: Leaving directory '/home/user/libimobiledevice-master'

Once I did that, I changed the permissions on the tools folder:
    sudo chmod 0777 -R tools/

I then paired the phone, and started the service:

    [user@localhost tools]$ sudo ./idevicebackup2 backup ~
    Backup directory is "/home/user"
    Started "com.apple.mobilebackup2" service on port 61791.
    Negotiated Protocol Version 2.1
    Starting backup...
    Backup will be encrypted.
    Requesting backup from device...
    Full backup mode.

I don't think the dependencies are the problem in this case. It may be the actual libimobiledevice package itself.

aburgh, 2015-06-29:
Did it stop at "Full backup mode." or did it do the backup? If it didn't run completely, can you run it again with `-d` now that you have the communication debugging enabled?

wputnam, 2015-06-30:
No, I just canceled it before it started backing up.
I used:
    sudo ./idevicebackup2 -d backup --full ~ > out.txt

The only information that echoed was what appears to be core dump information. Example:
    0000: 43 46 41 36 4c 50 41 41 4c 00 00 00 00 00 00 00   | CFA6LPAAL.......
    0010: 4c 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00   | L...............
    0020: 0d 00 00 00 00 00 00 00                           | ........
    
    0000: 02 00 00 00 00 00 00 00 2f 63 6f 6d 2e 61 70 70   | ......../com.app
    0010: 6c 65 2e 69 74 75 6e 65 73 2e 6c 6f 63 6b 5f 73   | le.itunes.lock_s
    0020: 79 6e 63 00                                       | ync.
    
    0000: 01 00 00 00 00 00 00 00                           | ........

The text file was almost 800 KB. From the entered command to the start of the sensitive data export:

    Backup directory is "/home/user"
    15:46:08 lockdown.c:654 lockdownd_client_new(): device udid: 7eac1b0047aba46808f95950443c71f95799b933
    15:46:08 lockdown.c:384 lockdownd_query_type(): called
    15:46:08 property_list_service.c:127 internal_plist_send(): sending 287 bytes
    15:46:08 service.c:140 service_send(): sending 4 bytes
    15:46:08 service.c:140 service_send(): sending 287 bytes
    15:46:08 property_list_service.c:132 internal_plist_send(): sent 287 bytes
    15:46:08 property_list_service.c:133 internal_plist_send(): printing 287 bytes plist:
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
    	<key>Label</key>
    	<string>idevicebackup2</string>
    	<key>Request</key>
    	<string>QueryType</string>
    </dict>
    </plist>
    15:46:08 property_list_service.c:191 internal_plist_receive_timeout(): initial read=4
    15:46:08 property_list_service.c:200 internal_plist_receive_timeout(): 297 bytes following
    15:46:08 property_list_service.c:213 internal_plist_receive_timeout(): received 297 bytes
    15:46:08 property_list_service.c:239 internal_plist_receive_timeout(): printing 297 bytes plist:
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
    	<key>Request</key>
    	<string>QueryType</string>
    	<key>Type</key>
    	<string>com.apple.mobile.lockdown</string>
    </dict>
    </plist>
    15:46:08 lockdown.c:400 lockdownd_query_type(): success with type com.apple.mobile.lockdown
    15:46:08 userpref.c:170 userpref_get_config_dir(): initialized config_dir to /var/lib/lockdown
    15:46:08 property_list_service.c:127 internal_plist_send(): sending 4989 bytes
    15:46:08 service.c:140 service_send(): sending 4 bytes
    15:46:08 service.c:140 service_send(): sending 4989 bytes
    15:46:08 property_list_service.c:132 internal_plist_send(): sent 4989 bytes
    15:46:08 property_list_service.c:133 internal_plist_send(): printing 4989 bytes plist:
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">

I'm looking through the log and I don't see any blatant transmission errors. Is there anything important I should be on the lookout for?

aburgh, 2015-06-30:
> I'm looking through the log and I don't see any blatant transmission errors. Is there anything important I should be on the lookout for?

If I understand correctly, you are saying the build from source works as expected and only stopped because you canceled it. In that case there aren't any transmission errors to find. If your build from github works, doesn't that point to the Fedora package being out of date?

> The only information that echoed was what appears to be core dump information.

It's not a core dump, it's a standard tag (I don't recall where it's used, but I know I've seen it).

wputnam, 2015-06-30:
Yes, that is what I mean.

I've checked the version installed by Fedora, and it's libimobiledevice-1.2.0-1.fc22.x86_64. According to the website for libimobiledevice, that's the latest stable version.

aburgh, 2015-07-02:
What's the upshot, that the Fedora maintainers need to update the package for libimobiledevice?

kev40293, 2015-07-09:
I've seen this issue on both ubuntu 14.04, and fedora 22 (will test on 15.04 soon)., and if i extract the package from the src.rpm and build, it works fine, and It works out of the box on arch.  Could it be something weird related to splitting the tools and library?

kev40293, 2015-07-09:
i was able to reproduce the error in the release tarball with
./configure --disable-static
./tools/idevicebackup backup ~/ipad

aburgh, 2015-07-09:
Since you disabled static libraries and did not install your built products, would `./tools/idevicebackup` link against the .so installed by the Fedora package, not the .so you built?

kev40293, 2015-07-09:
well i thought of that, so i also did
LD_LIBRARY_PATH=../src/.libs ./idevicebackup2... 
and used ldd with the same prefix to verify that it was using the so that I built. Same error occurs.
Also, i reached this point because the fedora package compiles with
./configure --disable-static --disable-openssl --enable-dev-tools
(though oddly removing the disable static and rebuilding the rpm doesnt fix it either)

UnitedMarsupials, 2016-02-21:
I'm getting this error on FreeBSD right now... It kept giving me this error until I tried rebooting the phone. The reboot helped and I saw the `idevicebackup2` proceed once after that. The utility core-dumped shortly afterwards, however, and is now issuing the same error again:

    % idevicebackup2 backup --full -d /home/mi/iPhone/
    Backup directory is "/home/mi/iPhone/"
    ERROR: Could not start service com.apple.mobilebackup2.

After one more reboot and using `idevicebackup2` recompiled for debugging (no optimization and `-g`) I was able to perform a backup. Well, I think, I was able too: after running to 100%  and saying:

    Sending '84e18d3a732...4c962ebb99937ce7/Manifest.mbdb' (1.5 MB)

the utility now hangs (going from "select" to "nanslp" state and back). In the backup directory there are thousands of files with unreadable names. The three last-updated files are:

    -rw-r--r--  1 root  wheel      24026 21 лют 14:32 Manifest.plist
    -rw-r--r--  1 root  wheel    1514126 21 лют 14:32 Manifest.mbdb
    -rw-r--r--  1 root  wheel        189 21 лют 14:32 Status.plist

The timestamps -- 14:32 -- is 26 minutes ago. At that same moment the iPhone's own syslog said:

    Feb 21 14:32:10 Misa BackupAgent2[191] <Warning>: INFO: Backup properties: <MBDriveProperties: version=9.1, systemDomainsVersion=24.0, date=2012-21 19:32:04 +0000, encrypted=1, passcodeSet=0, lockdownKeys={com.apple.MobileDeviceCrashCopy: {}, com.apple.TerminalFlashr: {}, com.apple.mob.data_sync: {Bookmarks: {AccountNames: [], Sources: []}, Contacts: {AccountNames: ["Exchange"], Sources: ["com.apple.dataaccess.EAS"]}, Calenda {AccountNames: [], Sources: ["com.apple.dataaccess.EAS"]}, Notes: {AccountNames: ["aldan.algebra.com"], Sources: ["com.apple.dataaccess.IMAPNo"]}}, com.apple.Accessibility: {MonoAudioEnabledByiTunes: 0, VoiceOverTouchEnabledByiTunes: 0, ZoomTouchEnabledByiTunes: 0, SpeakAutoCorrectionabledByiTunes: 0, InvertDisplayEnabledByiTunes: 0, ClosedCaptioningEnabledByiTunes: 0}, ProductVersion: "9.2.1", ProductType: "iPhone6,1", Builrsion: "13D15", com.apple.mobile.iTunes.accessories: {}, com.apple.mobile.wireless_lockdown: {}, UniqueDeviceID: "84e18d3a732b493cd09fe3dc4c96299937ce7", SerialNumber: "F17LQ2UAFNJT", DeviceName: "ÐÑÑÐ°"}>

Attaching a debugger to the running `idevicebackup2` process, I see the following two threads:

    (gdb) info threads
      2 Thread 802c06400 (LWP 101299/idevicebackup2)  0x000000080180398a in _sendto () from /lib/libc.so.7
    * 1 Thread 802c07000 (LWP 101618/idevicebackup2)  0x00000008017c0b5a in _nanosleep () from /lib/libc.so.7

Thread 1 is not interesting -- `np_notifier` sleeping -- but thread 2 gives the following backtrace:

     (gdb) bt full
     #0  0x000000080180398a in _sendto () from /lib/libc.so.7
     No symbol table info available.
     #1  0x00000008014d20ef in pthread_suspend_all_np () from /lib/libthr.so.3
     No symbol table info available.
     #2  0x00000008010ba835 in usbmuxd_send () from /opt/lib/libusbmuxd.so.4
     No symbol table info available.
     #3  0x000000080083770f in internal_connection_send (connection=0x802c4d780, data=0x7fffffff5c70 "\001У", len=32768, sent_bytes=0x7fffffff5ab0) at idevice.c:330
        res = 0
     #4  0x000000080083768d in idevice_connection_send (connection=0x802c4d780, data=0x7fffffff5c70 "\001У", len=32768, sent_bytes=0x7fffffff5ab0) at idevice.c:363
     No locals.
     #5  0x00000008008383f0 in service_send (client=0x802c1f2b0, data=0x7fffffff5c70 "\001У", size=32768, sent=0x7fffffff5afc) at service.c:141
        res = SERVICE_E_UNKNOWN_ERROR
        bytes = 0
     #6  0x000000080084876d in mobilebackup2_send_raw (client=0x802c1f1e8, data=0x7fffffff5c70 "\001У", length=32768, bytes=0x7fffffffdc84) at mobilebackup2.c:228
        raw = 0x802c1f2b0
        bytes_loc = 0
        sent = 0
     #7  0x0000000000409bf7 in mb2_handle_send_file (mobilebackup2=0x802c1f1e8, backup_dir=0x7fffffffe758 "/home/mi/iPhone/", 
    path=0x802c16140 "84e18d3a732...4c962ebb99937ce7/Manifest.mbdb", errplist=0x7fffffffdd00) at idevicebackup2.c:564
        r = 32768
        nlen = 25165824
        pathlen = 54
        bytes = 0
        localfile = 0x802c1e420 "/home/mi/iPhone//84e18d3a732...4c962ebb99937ce7/Manifest.mbdb"
        buf = "\001У\000\000\001УT\216\036vUСqET\216\036v\000\000\000\000\000\000\000\000\003\000\000\vMediaDomain\000\\Library/SMS/Attachments/dd/13/470AABCF-332D-414C-9AC1-0688AF08E256/IMG952014040495151954.jpgЪЪЪЪ\000,\003\000\000\000╙\031\t`ґE_:╧Z\001hґ\207DЪгеУщ╝\n0EZ\"Шф1Э╧Ґї{\037\224oF╦V\201є\000\000\000\000\000\000`Q\000\000\001У\000"...
        fst = {st_dev = 4146610099, st_ino = 3762693, st_mode = 33188, st_nlink = 1, st_uid = 0, st_gid = 0, st_rdev = 4294967295, st_atim = {tv_sec = 1456083130, 
    tv_nsec = 116735348}, st_mtim = {tv_sec = 1456083130, tv_nsec = 161730317}, st_ctim = {tv_sec = 1456083130, tv_nsec = 879731742}, st_size = 1514126, st_blocks = 1, st_blksize = 131072, st_flags = 2048, st_gen = 8013408, st_lspare = 0, st_birthtim = {tv_sec = 1456083130, tv_nsec = 116735348}}
        f = (FILE *) 0x801a84860
        slen = 0
        errcode = -1
        result = -1
        length = 32768
        total = 1514126
        sent = 393216
        err = MOBILEBACKUP2_E_SUCCESS
        format_size = 0x802c1f0b8 "1.5 MB"
     #8  0x0000000000407f43 in mb2_handle_send_files (mobilebackup2=0x802c1f1e8, message=0x802c42920, backup_dir=0x7fffffffe758 "/home/mi/iPhone/") at idevicebackup2.c:638
[... cut ...]
- `idevicebackup2` is not working quite right -- backing up with iTunes has never shown this sort of error with this same phone, nor has it ever hung at the end for me either.
- There is a bug in the utility, which depends on compiler optimization (or lack thereof). One possible reason for the problem is that the device is named in Cyrillic: "Міша" (my name in Ukrainian). The phone uses the transliteration of that ("Misa") internally, but, maybe, something inside libimobiledevice and friends is not coping with unexpected characters quite right?

zgoda-mobica, 2016-04-26:
I have the same error on Ubuntu 16.04, in syslog:

```
Apr 26 17:41:18 iPhone afcd[225] <Error>: Could not check in with lockdown at 'xpc': 1
Apr 26 17:41:18 iPhone com.apple.xpc.launchd[1] (com.apple.afcd[225]) <Warning>: Service exited with abnormal code: 1
Apr 26 17:41:18 iPhone BackupAgent2[226] <Error>: Could not check in with lockdown at 'xpc': 1
```

and in terminal:

```
$ idevicebackup2 backup -d --full iphone_bkp
Backup directory is "iphone_bkp"
ERROR: Could not start service com.apple.mobilebackup2.
```

FYI:

```
$ apt-cache show libimobiledevice-utils 
Package: libimobiledevice-utils
Priority: optional
Section: universe/utils
Installed-Size: 378
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: gtkpod Maintainers <pkg-gtkpod-devel@lists.alioth.debian.org>
Architecture: amd64
Source: libimobiledevice
Version: 1.2.0+dfsg-2.1build1
Depends: libc6 (>= 2.15), libgcrypt20 (>= 1.6.1), libgnutls30 (>= 3.4.0), libimobiledevice6 (>= 1.2.0), libplist3 (>= 1.11), libtasn1-6 (>= 4.5), libusbmuxd4 (>= 1.0.10)
Filename: pool/universe/libi/libimobiledevice/libimobiledevice-utils_1.2.0+dfsg-2.1build1_amd64.deb
Size: 80768
MD5sum: 9ac2f334e4fdc48d7b624bda97dfb35e
SHA1: 0ae3adb5dc6c98521a1ddcf8cde1fad185196b5f
SHA256: a338a8419f9c53a11039f58b886fa6a704b14d5d099d8f25691286077462637f
Description-en: Library for communicating with iPhone and iPod Touch devices
 libimobiledevice is a library that talks the native Apple USB protocols that
 the iPhone and iPod Touch use. Unlike other projects, libimobiledevice does
 not depend on using any existing libraries from Apple.
 .
 This package contains utilities and examples which use libimobiledevice.
Description-md5: 67c848cadcb0e4665fbbe126820e4bd8
Homepage: http://libimobiledevice.org/
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Origin: Ubuntu
Supported: 3y
Task: ubuntu-mate-desktop
```

zgoda-mobica, 2016-04-27:
And more from system log:

```
Apr 26 22:52:30 ubuntu kernel: [ 2567.073658] upowerd[32537]: segfault at 0 ip 00007f70c080c9a0 sp 00007ffeba862030 error 4 in libimobiledevice.so.6.0.0[7f70c0801000+28000]
Apr 26 22:52:30 ubuntu systemd[1]: upower.service: Main process exited, code=dumped, status=11/SEGV
```

inflex, 2016-05-14:
@zgoda-mobica same problem here, trying to get it working with 16.04, also tried some older releases and still the same.  What's most painful is that it *was* working with my previous 15.10 setup.

I did a full strip out of all the U16.04 supplied items and rebuilt from source and in to /usr/local/ but still no success.

Tagging myself in to this thread so I can see if there's a solution.

(note, I found "ideviceinfo" fails with lockdownd -5, but "ideviceinfo -s" works, the difference being, apparently that -s doesn't utilise SSL comms), not sure if it's related but thought I'd throw that in.

zgoda-mobica, 2016-05-24:
This is somehow related to Ubuntu/Debian packaging. Everything works fine if I build libimobiledevice and tools from raw source, fails if I build from Ubuntu src.deb (and on Debian Jessie too).

macplox, 2017-07-06:
I don't think it's a Debian thing. I built from source and I'm able to use it to backup an ipod touch 3g with idevicebackup, but i cannot backup and iphone 5s ios 10.3.2 with idevicebackup2 on the same machine.

LetsUnlockiPhone, 2017-09-28:
com.apple.mobilebackup
available only after device receives activation from albert.apple.com 
make sure you have passed activation 
I compiled on Fedora, Ubuntu, Mac and Win and all working perfectly for all IOS versions

nikias, 2017-09-28:
Guys can you make sure you are using the latest code from git? The pre-compiled packages from the distributions are outdated. I can use the latest code to back up even on iOS 11 without any issues.

nikias, 2022-03-28:
Not a bug in libimobiledevice.

closed 2022-03-28 by nikias