libimobiledevice/libimobiledevice #208: Ubuntu PPA for 1.2 version ?

lakano, 2015-05-13:
Hello,

I don't found clean solution to install libimobiledevice 1.2 on my Ubuntu Vivid.
I see it's in experimental inside Debian.

Do you known if someone have made a PPA to install it properly ? 

Thanks a lots ! :-)

( PS : I need the 1.2 version to use ios-webkit-debug-proxy )

octoxan, 2015-09-11:
I also need the 1.2 version on 12.04 to use ios-webkit-debug-proxy

Vinnipeg, 2015-09-12:
+

tuliomarchetto, 2015-09-12:
I also need a 1.2 PPA

frankxu2004, 2015-10-13:
After a quick search, https://launchpad.net/~martin-salbaba/+archive/ubuntu/ppa+libimobiledevice seems like a updated PPA.

Yanpas, 2015-10-13:
@frankxu2004 thanks! But that didn't solved my problem, still no Application Data directory since iOS 8 #244 
Maybe not all components have been updated in this ppa

mathieujobin, 2019-01-09:
a ubuntu PPA with builds from latest from HEAD would be most helpful.

I'd like to see imobiledevice working on Ubuntu 18.04 and iOS 12

Thanks