libimobiledevice/libimobiledevice #209: No way to clear console log with idevicesyslog

rgonzo, 2015-05-15:
There is currently no way to clear the console log with the command line tool - idevicesyslog. 

When using a tool like Xcode, when I have the option to clear the log with the following button: 

[image]

Is there any way we can pass in an argument to the idevicesyslog command that will clear the console?

nikias, 2015-05-23:
idevicesyslog is a console application and I don't think there is a portable way to implement a "clear console" feature.
On OS X when using the Terminal.app you can use ALT+CMD+R to do a "Hard Reset".

MishaZhe, 2015-08-20:
Hi,

We can collect logs from some period of time using workaround -  
start writing logs using "libimobiledevice" into some file:

    $ idevicesyslog > tempLogs.txt

then from another entry call tail command:

    $ tail -f tempLogs.txt > LogsFromExactPeriodOfTime.txt

I believe something like this can be integrated to "libimobiledevice", if it's already not.

Cheers,
Misha