libimobiledevice/libimobiledevice #210: idevicedebug very slow logging

nchapp, 2015-05-21:
This may already be a known issue, but I have found that running and debugging an app using idevicedebug is not practical because of how slowly it runs. Currently if I run an app using idevicedebug the logs I see are are always extremely delayed from what is actually occurring on the device.

AntonShpk, 2016-10-13:
look at 
tools/idevicedebug.c 
line 467: sleep(1)

dolphineye, 2018-11-16:
Hi, we have the same issue here. Any update on this topic? Regards!

oddgames-david, 2019-02-27:
Also experiencing this

nikias, 2019-02-27:
@AntonShpk the sleep is in another thread, this just waits until devices appear/disappear via the registered callback handler, or ctrl+c is pressed.

I can't confirm this on my end. The data is printed as it arrives, additionally flushed to stdout whenever a linebreak is received. There's no guarantee the device sends the log output without any delays, but for me it is and has always been pretty instant.

What platform are you guys on?

oddgames-david, 2019-02-27:
@nikias Windows 10, tried with iPhone 5 iOS 10.3.3 and iPhone 6S iOS 11.4.1

Running the app from Xcode gives me instant logs but ilibimobiledevice is very delayed, can take a few minutes to get past the start of my app which should print out in a 1-3 seconds.

Heres what it looks like (there is some legitimate delay between a couple of the lines but it's nowhere near whats' shown).

[image]

nikias, 2019-02-27:
@oddgames-david what about idevicesyslog on macOS? Windows has always been strange...

oddgames-david, 2019-02-27:
@nikias idevicesyslog performs perfectly on Windows but it doesn't give me any useful logs from my Unity app, it's all just input, wifi, battery levels etc.

I've not tried any of the tools on osx, just output from xcode.

nikias, 2019-02-27:
Ohh I totally misread idevicedebug for idevicesyslog. Now it makes sense.

nikias, 2019-02-27:
Change the sleep(1) to something like usleep(10000), or Sleep(10) on windows.

oddgames-david, 2019-02-27:
Ok, I guess it's time to leave the safety of my C# skillset and experience some C 👍  I think I see the problem with x*1000

nikias, 2019-02-27:
The problem is not the x*1000, but the sleep in general being too long with 1 second

oddgames-david, 2019-02-27:
@nikias It only took my 7 hours but I finally got something building with that sleep changed to a lower value and it definitely presents the characters that I was seeing previously faster but it still pauses in the same spots half way between messages.

oddgames-david, 2019-02-28:
idevicedebug has the same speed on osx (installed latest via homebrew)

[image]

nikias, 2019-02-28:
Yes I know. As I said I confused it for idevicesyslog. I think it needs another thread. Will try to look at it tomorrow...

oddgames-david, 2019-04-11:
Any progress on this one?

Claytonious, 2019-05-29:
Is there any news on this problem?

anton-kl, 2020-11-03:
One workaround is to run the app with `OS_ACTIVITY_DT_MODE=enable` like this:
`idevicedebug -e OS_ACTIVITY_DT_MODE=enable run com.my.app`
This will ensure NSLog, CFLog and os_log messages get mirrored to stderr (see https://github.com/ios-control/ios-deploy/blob/master/src/scripts/lldb.py#L55)
Then you can use `idevicesyslog` to get logs from device, which will include logs from your app - in real time.