libimobiledevice/libimobiledevice #215: house_arrest_get_result(...) = HOUSE_ARREST_E_CONN_FAILED (-3) with simple application.

theonlylawislove, 2015-05-31:
I compiled libimobiledevice (Win32) and I am trying to run a minimal example just to play with things.

My source is very simple.

```
#include "stdafx.h"

#include <libimobiledevice/libimobiledevice.h>
#include <libimobiledevice/lockdown.h>
#include <libimobiledevice/afc.h>
#include <libimobiledevice/house_arrest.h>

#define AFC_SERVICE_NAME "com.apple.afc"

idevice_t phone = NULL;
lockdownd_client_t control = NULL;
lockdownd_service_descriptor_t service = NULL;
house_arrest_client_t house_arrest = NULL;

int _tmain(int argc, _TCHAR* argv[])
{
	int res = EXIT_FAILURE;
	lockdownd_error_t lockdownResult = LOCKDOWN_E_SUCCESS;
	house_arrest_error_t houseArrestResult = HOUSE_ARREST_E_SUCCESS;

	idevice_new(&phone, NULL);

	lockdownResult = lockdownd_client_new_with_handshake(phone, &control, "ifuse");

	lockdownResult = lockdownd_start_service(control, AFC_SERVICE_NAME, &service);

	houseArrestResult = house_arrest_client_new(phone, service, &house_arrest);

	if (!house_arrest) {
		fprintf(stderr, "Could not start document sharing service!\n");
		return EXIT_FAILURE;
	}

	houseArrestResult = house_arrest_send_command(house_arrest, "VendDocuments", "com.adobe.Adobe-Reader");

	plist_t dict = NULL;
	houseArrestResult = house_arrest_get_result(house_arrest, &dict);

	return 0;
}
```

I have checked on the return results, and everything passes/succeeds example the very last line, which I get a ```HOUSE_ARREST_E_CONN_FAILED (-3)``` error.

I took this code exactly (minus error checking, etc) from ifuse (https://github.com/libimobiledevice/ifuse/blob/master/src/ifuse.c).

Why am I getting a connection failed? I am using iOS 8.3, and it is my understanding that "VendDocuments" works with this version.

Here is the entire output from my console application.

```
                aWNkUU9iVWVvQ0pmemxxQmg3TzhYRjVqMkZBcEtLOW5HUWxnVkJXUkpkcGYwMgpH
V0ho
                a1NtQkxxckJjd0FHdUtDYXZBUHpIanQySWc5K3UwYlMxNVlFV1ExT3RQdkhveXBz
cGll
                Tm4rcGFOd2JRCm1DVUs0ZEZiOTd3WURPNnhwV3lYQllSei9nbVkzMGk1VEZxVjR5
RmJw
                VW1iTnhSaTIzYzc3YUs0azJjVWE1YTAKa2NlTmtLdWxjQS9ZM1gxWEVxWDNUMmZW
bVJN
                Q0F3RUFBYU0vTUQwd0RBWURWUjBUQVFIL0JBSXdBREFkQmdOVgpIUTRFRmdRVTE1
KzE5
                ZW5vRC9kL21ycjdOVHZ2NEszUDRpZ3dEZ1lEVlIwUEFRSC9CQVFEQWdXZ01BMEdD
U3FH
                ClNJYjNEUUVCQlFVQUE0SUJBUUREYlEzRnlhVUZlY0ZUUU1zVzNxRDV6UCtPN2Zr
d2tG
[... 261 lines cut ...]
<plist version="1.0">
<dict>
        <key>Command</key>
        <string>VendDocuments</string>
        <key>Identifier</key>
        <string>com.adobe.Adobe-Reader</string>
</dict>
</plist>
16:52:30 [0x2ef0] ->usbmuxd_recv_timeout(612, 4)
16:52:30 [0x2ef0] <-usbmuxd_recv_timeout(612, 4)
16:52:30 [0x2ef0] d:\git\libimobiledevice-windows-build\libimobiledevice-source\
src\idevice.c:381 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_tim
eout returned -11 (Resource temporarily unavailable)
16:52:30 [0x2ef0] d:\git\libimobiledevice-windows-build\libimobiledevice-source\
src\service.c:163 service_receive_with_timeout(): could not read data
16:52:30 [0x2ef0] d:\git\libimobiledevice-windows-build\libimobiledevice-source\
src\property_list_service.c:191 internal_plist_receive_timeout(): initial read=0

16:52:30 [0x2ef0] d:\git\libimobiledevice-windows-build\libimobiledevice-source\
src\property_list_service.c:193 internal_plist_receive_timeout(): initial read f
ailed!
16:52:30 [0x2ef0] d:\git\libimobiledevice-windows-build\libimobiledevice-source\
src\house_arrest.c:141 house_arrest_get_result(): could not get result, error -3

```

theonlylawislove, 2015-05-31:
I think I found the issue.

I need to be using ```HOUSE_ARREST_SERVICE_NAME``` (com.apple.mobile.house_arrest) instead of ```AFC_SERVICE_NAME```.

mexmer, 2015-05-31:
HA service and AFC service are two different services.

HA is for accessing application sandbox.

AFC is regular filesystem access.

theonlylawislove, 2015-05-31:
So, should I perform the house arrest first, and then perform all the AFC methods? After I do all the house arrest methods, the AFC methods will be mounted at the application directory that I sandboxed?

theonlylawislove, 2015-05-31:
Is there a source file somewhere that shows the exact sequence of events to establish a connection to an iOS device, sandbox an app by bundle identifier, and perform mkdir (or some other AFC method) in the applications "Documents" directory? I am reading the API and I feel like I am putting a puzzle together.

mexmer, 2015-05-31:
HA service is quite special. 
start service 
send command (venddocumments)
if you receive succesfull response, use socket that you used for HA service, as socket for afc client

theonlylawislove, 2015-05-31:
Great. I know see that ```afc_client_new_from_house_arrest_client``` is what I need. I am able to call this method with success. But now, when I call ```afc_make_directory```, I get ```AFC_E_MUX_ERROR (30)```.

Here is the code. There is no error checks, but in the debugger, everything passes, exect the last line.

```
int _tmain(int argc, _TCHAR* argv[])
{
	idevice_event_subscribe(&device_subscribed, NULL);

	int res = EXIT_FAILURE;
	lockdownd_error_t lockdownResult = LOCKDOWN_E_SUCCESS;
	house_arrest_error_t houseArrestResult = HOUSE_ARREST_E_SUCCESS;
	afc_error_t afcResult = AFC_E_SUCCESS;

	idevice_new(&phone, NULL);

	lockdownResult = lockdownd_client_new_with_handshake(phone, &control, "ifuse");

	lockdownResult = lockdownd_start_service(control, HOUSE_ARREST_SERVICE_NAME, &service);

	houseArrestResult = house_arrest_client_new(phone, service, &house_arrest);

	if (!house_arrest) {
		fprintf(stderr, "Could not start document sharing service!\n");
		return EXIT_FAILURE;
	}

	houseArrestResult = house_arrest_send_command(house_arrest, "VendDocuments", "com.adobe.Adobe-Reader");

	plist_t dict = NULL;
	houseArrestResult = house_arrest_get_result(house_arrest, &dict);

	plist_t node = plist_dict_get_item(dict, "Error");
	if (node) {
		char *str = NULL;
		plist_get_string_val(node, &str);
		fprintf(stderr, "ERROR: %s\n", str);
		if (str) free(str);
		return EXIT_FAILURE;
	}
	plist_free(dict);

	afc_client_t afc = NULL;
	afcResult = afc_client_new_from_house_arrest_client(house_arrest, &afc);

	afcResult = afc_make_directory(afc, "/Documents/test");

	return 0;
}
```

Any idea why ```afc_make_directory``` is failing?

theonlylawislove, 2015-06-01:
After looking more into it, it seems that the error code ```-34``` is not the correct error code. On win platforms, you have to get the error number using ```WSAGetLastError```. After changing the code, the error number returned is WSAECONNABORTED (10053). This error code means the following.

> Software caused connection abort.
An established connection was aborted by the software in your host computer, possibly due to a data transmission time-out or protocol error.

Any ideas on the causes of this?

mexmer, 2015-06-01:
i'm not sure if create folder operation is supported in houserarrest at all.
so far i have confirmed working only readfile, writefile, createfile, lockfile, deletefile.

theonlylawislove, 2015-06-01:
I got it working!

Turns out that the issue was the byte order. I didn't have the __LITTLE_ENDIAN__ preprocessor defined.

Also, mkdir works.

closed 2015-06-01 by theonlylawislove

aburgh, 2015-06-01:
I can confirm that you can create and delete directories. I have a fork of libimobiledevice which has another tool called `afccl` that has a few shell commands for interacting with files via AFC, and I can create and delete a test directory in Good Reader with it. See http://github.com/aburgh/libimobiledevice for example code. I only use in on OS X and it depends on readline, but it's useful for reference.

mexmer, 2015-06-01:
that's good news. i know, it was working on previous version of OS, i used it for some app HA, but have not tested it much on 8.3 ... there were lot of changes due security ... but no time to test all of them.