libimobiledevice/libimobiledevice #216: heap buffer overflow in debugserver.c

progressnerd, 2015-06-02:
We've found memory stomping issues in the libimobiledevice debugserver. ASan yields the output printed below. It appears as though debugserver_decode_string is being called on responses that lack a hexadecimal string, including the simple response "OK".

==18867==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000022a51 at pc 0x0000004dc000 bp 0x7fff1e3982d0 sp 0x7fff1e3982c8
WRITE of size 1 at 0x602000022a51 thread T0
    #0 0x4dbfff in debugserver_decode_string third_party/libimobiledevice/src/debugserver.c:285:5
    #1 0x4d9eb8 in debugserver_client_handle_response third_party/libimobiledevice/tools/idevicedebug.c:112:3
    #2 0x4d9346 in main third_party/libimobiledevice/tools/idevicedebug.c:477:13
    #3 0x7fa185548ce7 in __libc_start_main (/usr/grte/v4/lib64/libc.so.6+0x38ce7)
    #4 0x423d48 in _start /usr/grte/v4/debug-src/glibc-2.19/csu/../sysdeps/x86_64/start.S:118

0x602000022a51 is located 0 bytes to the right of 1-byte region [0x602000022a50,0x602000022a51)
allocated by thread T0 here:
    #0 0x4c4aab in __interceptor_malloc third_party/llvm/llvm/projects/compiler-rt/lib/asan/asan_malloc_linux.cc:40:3
    #1 0x4dbee5 in debugserver_decode_string third_party/libimobiledevice/src/debugserver.c:277:12
    #2 0x4d9eb8 in debugserver_client_handle_response third_party/libimobiledevice/tools/idevicedebug.c:112:3
    #3 0x4d9346 in main third_party/libimobiledevice/tools/idevicedebug.c:477:13
    #4 0x7fa185548ce7 in __libc_start_main (/usr/grte/v4/lib64/libc.so.6+0x38ce7)
    #5 0x423d48 in _start /usr/grte/v4/debug-src/glibc-2.19/csu/../sysdeps/x86_64/start.S:118

SUMMARY: AddressSanitizer: heap-buffer-overflow third_party/libimobiledevice/src/debugserver.c:285:5 in debugserver_decode_string
Shadow bytes around the buggy address:
  0x0c047fffc4f0: fa fa 00 00 fa fa fd fd fa fa fd fd fa fa 00 00
  0x0c047fffc500: fa fa fd fd fa fa fd fd fa fa 00 00 fa fa fd fd
  0x0c047fffc510: fa fa fd fd fa fa 00 00 fa fa fd fd fa fa fd fd
  0x0c047fffc520: fa fa 00 00 fa fa fd fd fa fa fd fd fa fa 00 00
  0x0c047fffc530: fa fa 00 00 fa fa fd fa fa fa fd fd fa fa fd fd
=>0x0c047fffc540: fa fa fd fd fa fa 00 00 fa fa[01]fa fa fa 00 07
  0x0c047fffc550: fa fa fd fd fa fa fd fd fa fa fd fd fa fa fd fd
  0x0c047fffc560: fa fa fd fd fa fa fd fd fa fa fd fd fa fa fd fd
  0x0c047fffc570: fa fa fd fd fa fa fd fd fa fa fd fd fa fa fd fd
  0x0c047fffc580: fa fa fd fd fa fa fd fd fa fa fd fd fa fa fd fd
  0x0c047fffc590: fa fa 03 fa fa fa fd fd fa fa fd fd fa fa fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00

progressnerd, 2015-06-02:
Seems this issue can be fixed by wrapping the call to `debugserver_client_handle_response` on [Line 462](https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicedebug.c#L462) with the guard with `if (strncmp(response, "OK", 2))` in the same way other calls to this function are guarded.

nikias, 2022-03-29:
Fixed in git

closed 2022-03-29 by nikias