libimobiledevice/libimobiledevice #217: Turning off wifi on iPhone through command line tool

harish2304, 2015-06-17:
Can we turn off the wifi in iPhone by using libmobiledevice tool.Is it possible?
Anyhelp is appreciated.

Thanks
Harish

qianguozheng, 2015-06-25:
I believe it can. but how to do it, i have no idea

jevinskie, 2015-06-25:
idevicediagnostics shutdown

harish2304, 2015-06-26:
i don't want to shut down the device, just want to toggle wifi ..

mexmer, 2015-06-26:
there is no command to toggle wifi afaik
it's only possible trough settings (on device ui)

shermanericSF, 2015-09-01:
I created this.  It uses networksetup.  I know it could be cleaner.  hope it helps!
https://gist.github.com/shermanericSF/9dac47f0a8d804011d08

tarrenj, 2015-09-01:
shermanicSF,

So I'd have to run this on the device I wanted to change the network
setting on, right?  I can't connect my iDevice to a computer, and run this
on there to change the settings on the iDevice?

shermanericSF, 2015-09-01:
right.  This assumes you set up your host desktop up for internet sharing.  Then, on your device you pick this access point.  I would also forget other nearby devices.  The connection shaping happens downstream on the host instead of directly on the device, but if set up correctly has the same effect.

FunkyM, 2015-09-25:
Not really. There are ways to do it but nothing straightforward like an on/off command.

closed 2015-09-25 by FunkyM