libimobiledevice/libimobiledevice #218: Unable to detect 2 devices at the same time

ivanceras, 2015-06-17:
When I connect 2 devices, I can only access the first 1. If I pull the first device, that's the only time I can connect to the other device.

`idevice_id -l` shows only 1 device id as well, instead of 2 device id for each iphones

tarrenj, 2015-06-22:
I'm also having this issue.  I was able to get around it by using the -i flag, and using the hex form of the unique_chip_id.

ivanceras, 2015-06-22:
`idevice_id -l` list down all the connected iphones in mac, however in linux (ubuntu, mint) only display the first attached iphone.

tarrenj, 2015-06-22:
Are you able to call ideviceinfo on multiple devices when using the '-u' flag? It seems like our problems are somewhat connected.

qianguozheng, 2015-06-25:
You must specify the ID when connect two device to on PC.
eg: ideviceinstaller -u <YOUR UDID> <iOS Package to be installed>.ipa

Charyas, 2015-10-08:
you must connect with the special ECID or UDID

kasperk81, 2021-06-01:
six years later... i'm seeing the same problem with `idevice_id` program on linux @nikias 

### Repro

connect two iphones to linux machine

### Expected

 `idevice_id -l` lists both devices

### Actual

`idevice_id -l` only shows the first device

nikias, 2021-06-01:
`idevice_id` can only show what `usbmuxd` is seeing. So you need to check on `usbmuxd` to find out what is wrong. I suggest unplugging all devices, then manually run `sudo usbmuxd -f -v -U usbmux` in another terminal, and check if it prints any errors/warnings once you plug a second device.