libimobiledevice/libimobiledevice #220: iPhone ask for Trust/Don't Trust when cable is plugged in after the first time

jmbrenna, 2015-06-23:
This is a question and not necessarily a bug.

When I plug my iPhone 5S/iOS 8.3 in, the Trust/Don't Trust dialog appears, as expected, and I click trust, and everything works as expected.  The problem is when I unplug and replug the iPhone, it again asks for me to trust the computer.  This makes automatic tethering not work because I must interact with the phone each time.   Any thoughts?

I am using the master branch of code and am still seeing the problem.

qianguozheng, 2015-06-25:
There some possible reason to your question, 
First, the usbmuxd generate udid.plist /var/lib/lockdownd/*.plist not saved.
Second, iOS system may have timeout mechnism for that, if you unplug too frequently.
I ocassionally would encounter this, but not every time.

hyunmin90, 2015-07-28:
I had simillar issue as well. Some of the syslog will not come through

gediminasgu, 2015-08-21:
Same issue here. I didn't have /var/lib/lockdownd directory created, so I have created by my self. But don't see any files created there and "Trust" alert still comes. Using latest libimobiledevice from the trunk (21st of August)

LeJeko, 2015-09-13:
I'm using OpenWRT and I've managed a script to backup UUIDs files and restore it at reboot.
You can inspire for your work.
https://www.dropbox.com/s/ujtapth0ra7ahmp/lockdown.sh?dl=0