libimobiledevice/libimobiledevice #221: unable to extract component: iBEC all iOS 8.x versions

MRLOLKOPF, 2015-07-17:
Hello!

I worked today first time with the idevicerestore tool on my iPhone 4S with my Windows 8.1 computer. I get this message when using a CUSTOM / OWN .ipsw.

Extracting iBEC.n94ap.RELEASE.dfu...
ERROR: zip_name_locate: Firmware/dfu/iBEC.n94ap.RELEASE.dfu
ERROR: Unable to extract iBEC.n94ap.RELEASE.dfu from C:\Users\Andreas\Desktop\iPhone4,1_8.4_12H143_Restore.ipsw
ERROR: Unable to extract component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

If i just recover with the "-l" option, it works. I tried several .ipsw files. iOS 8.4, 8.3, 8.1, 8.0... Nothing changed in them. They're original.

Thanks in advance!

SA-1, 2016-01-14:
@MRLOLKOPF I have the same issue, Did You get any fix ?
Please Reply !

Thanks in Advance.

MRLOLKOPF, 2016-01-14:
I'm sorry. I'm not sure, but I think I have solved my problem with my iPhone in some different way.

SA-1, 2016-01-14:
@MRLOLKOPF will You Please tell me how do you solve your problem ? 
as I am stuck at ibec. I tried hard but failed, also I have tried to install idevicerestore on mac but no success :(

MRLOLKOPF, 2016-01-14:
Like I said, I repaired my iPhone somehow else. I didn't used this tool for this. Sorry.

SA-1, 2016-01-14:
@MRLOLKOPF Ok Thanks !

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw