libimobiledevice/libimobiledevice #225: libimobiledevice build-error against GnuTLS 3.4

ametzler, 2015-07-25:
Hello,
libimobiledevice 1.2.0 fails to build against GnuTLS 3.4.0:

-------------------------
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..  -I../include -I..  -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden  -I/usr/include/p11-kit-1   -I/usr/include/libxml2  -D_FILE_OFFSET_BITS=64 -D_LARGE_FILES=1  -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -c -o idevice.lo idevice.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/include/p11-kit-1 -I/usr/include/libxml2 -D_FILE_OFFSET_BITS=64 -D_LARGE_FILES=1 -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -c idevice.c  -fPIC -DPIC -o .libs/idevice.o
idevice.c:629:172: error: unknown type name 'gnutls_retr_st'
 static int internal_cert_callback(gnutls_session_t session, const gnutls_datum_t * req_ca_rdn, int nreqs, const gnutls_pk_algorithm_t * sign_algos, int sign_algos_length, gnutls_retr_st * st)
                                                                                                                                                                            ^
idevice.c: In function 'idevice_connection_enable_ssl':
idevice.c:746:2: warning: implicit declaration of function 'gnutls_certificate_client_set_retrieve_function' [-Wimplicit-function-declaration]
  gnutls_certificate_client_set_retrieve_function(ssl_data_loc->certificate, internal_cert_callback);
  ^
idevice.c:746:77: error: 'internal_cert_callback' undeclared (first use in this function)
  gnutls_certificate_client_set_retrieve_function(ssl_data_loc->certificate, internal_cert_callback);
                                                                             ^
idevice.c:746:77: note: each undeclared identifier is reported only once for each function it appears in
Makefile:544: recipe for target 'idevice.lo' failed
make[3]: *** [idevice.lo] Error 1
make[3]: Leaving directory '/tmp/IMOB/libimobiledevice-1.2.0+dfsg/src'

---------------------------------------------

These functions were dropped in 3.4.0, see
http://www.gnutls.org/manual/html_node/Upgrading-from-previous-versions.html#Upgrading-from-previous-versions
for information on upgrading.
cu Andreas

ametzler, 2015-07-26:
Further info: Although the direct successor of gnutls_certificate_client_set_retrieve_function is gnutls_certificate_set_retrieve_function, and  callback’s function prototype is almost identical (it refernces gnutls_retr2_st* instead of gnutls_retr_st*) the GnuTLS documentation notes: "You are advised to use gnutls_certificate_set_retrieve_function2() because it is much more efficient in the processing it requires from gnutls."

FunkyM, 2015-09-25:
Thanks, this either misses a bump on the GnuTLS version in configure.ac or a way to compile with older versions.

kalev, 2015-10-05:
Sorry for the delay, I wasn't subscribed to the bug. Updated the pull request now to require gnutls 3.0.

nikias, 2016-04-29:
Closing. Fixed upstream supporting both the older and the newer API.

closed 2016-04-29 by nikias