libimobiledevice/libimobiledevice #226: ERROR: Could not connect to lockdownd, error code -8

markshep, 2015-08-25:
I'm writing a system to run on OSX which manages deploying app builds to iOS devices.  This uses a combination of tools from libmobiledevice and [ios-deploy](https://github.com/phonegap/ios-deploy).  I'm getting an issue where iOS devices seem to get into a bad state.  One symptom is the above error message when trying to use "ideviceinfo" and another is that deploying an app using either "ios-deploy" or Xcode never completes.

If I unplug and replug the device the error from "ideviceinfo" goes away and that tool then works fine.  However that doesn't fix the problem with deployment - any attempt to do so hangs and the -8 error with "ideviceinfo" comes back.  So it looks like the problem may not be with libimobiledevice itself, but as the only information I've got to go on is that error message this seems like the place to start!

Looking in the source the -8 code is LOCKDOWN_E_MUX_ERROR.  Undefining STRIP_DEBUG_CODE in config.h, rebuilding and running with "-d" gives the following:

    13:05:08 idevice.c:276 idevice_connect(): ERROR: Connecting to usbmuxd failed: -1 (Operation not permitted)
    13:05:08 lockdown.c:650 lockdownd_client_new(): could not connect to lockdownd (device 4e45f9d35a3edf7d5fb16bbc52d8354978faa2f9)
    13:05:08 lockdown.c:684 lockdownd_client_new_with_handshake(): failed to create lockdownd client.
    ERROR: Could not connect to lockdownd, error code -8

This is using the latest source in the "master" branch (commit f268393d4e447ac901879bee751d7350c495fed2).

Any suggestions as to what I should look at next to try and track down the exact cause of this problem?

tarrenj, 2015-08-25:
```
13:05:08 idevice.c:276 idevice_connect(): ERROR: Connecting to usbmuxd failed: -1 (Operation not permitted)
```

Are you running with sudo?

markshep, 2015-08-27:
No, should I be?

Grazfather, 2015-11-04:
I get this error if I plug the iphone into a new computer and don't click the 'allow' button on the phone screen, fwiw.

simonjgamer, 2016-05-18:
@markshep has this solution resolved for you? I am having similar error message

crummy, 2017-01-27:
I'm seeing this behaviour, with and without sudo, on some devices.

* OS X 10.11.5 (15F34)
* iPhone 5S and 5, both on iOS 9

nathan-fiscaletti, 2017-08-18:
I'm experiencing this error message when trying to run the tools with a device that's running iOS 11 Beta 6. Does this tool not yet support iOS 11?

nikias, 2017-08-18:
See #510 and #537. Golden rule for bug reporting: verify if it works with the latest version.

nathan-fiscaletti, 2017-08-19:
@nikias So, if this has been resolved why is the issue still marked as open?

nikias, 2017-08-20:
It's an old issue and it was obvious that it was a problem with usbmuxd. I didn't bother closing. But maybe now is the time.

closed 2017-08-20 by nikias