libimobiledevice/libimobiledevice #232: Store image in PNG Format

nijakanthan, 2015-09-10:
Actually this is not a issue. Is it possible to store the screenshot in png format

Thanks.

mexmer, 2015-09-10:
use conversion library (like imagemagick) to convert it to desired format. devices sends only one format.

nijakanthan, 2015-09-10:
Thanks @mexmer 
After adding imagemagick API can not run make command i am getting an error message.

Error message
Undefined symbols for architecture x86_64:
  "_DestroyMagickWand", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickGetException", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickNextImage", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickNormalizeImage", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickReadImage", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickRelinquishMemory", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickResetIterator", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickWandGenesis", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickWandTerminus", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_MagickWriteImages", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o
  "_NewMagickWand", referenced from:
      _main in idevicescreenshot-idevicescreenshot.o

Im new to MakeFile can anyone help me with this

nijakanthan, 2015-09-10:
Need to run below command to execute MagickWand API

gcc -o wandd wandd.c `pkg-config --cflags --libs MagickWand`

I can i do it using MakeFile

Thanks.

FunkyM, 2015-09-25:
As said, use external means to convert the screenshot to a different file format.
If you have improvements to the source code which adds more image file formats on the fly feel free to post a pull request and mind to share your code.

closed 2015-09-25 by FunkyM