libimobiledevice/libimobiledevice #234: How to backup all files of iphone  without  '/var/mobile/Media' folder ？

leuzz, 2015-09-18:
I do not need the images and videos in  '/var/mobile/Media' folder .  It takes a long time to backup the images and videos when i make a  full backup .  Is there other way to backup all files of iphone  without  '/var/mobile/Media' folder ？
Thank you !

nikias, 2015-09-28:
Since the device controls what gets backed up you normally can't change that behavior.

What you can try - **but at your own risk, the device might lose data, you have been warned** - is renaming the DCIM and PhotoData folders to something else like DCIM.bak and PhotoData.bak via AFC (e.g. using ifuse) before starting the backup. After the backup rename them to their original name. Do not reboot the device during the time you have the folder names changed.

Closing since this is neither a bug nor a (reasonable) feature request.

closed 2015-09-28 by nikias

leuzz, 2015-09-28:
Hello, nikias. Thank you very much for your answer ! I think it's a good idea .  I will have a try !