libimobiledevice/libimobiledevice #238: idevicebackup2 failed and the error si 'MOBILEBACKUP2_E_NO_COMMON_VERSION'

RonnieLLL, 2015-09-28:
I can not backup iPad mini with ios 8.1.2. on windows 7, And there had same problems, error "Could not perform backup protocol version exchange, error code -6" . 
the CMD: -u 5eee5e90e10ade52d5d7690fe5bf63aff60f5cd8 backup --full "C:\Users\Desktop\testbackup"

/*
double local_versions[2] = {2.0, 2.1};
double remote_version = 0.0;
err = mobilebackup2_version_exchange(mobilebackup2, local_versions, 2, &remote_version);
if (err != MOBILEBACKUP2_E_SUCCESS) {
printf("Could not perform backup protocol version exchange, error code %d\n", err);
cmd = CMD_LEAVE;
goto checkpoint;
}
*/
the 'err == MOBILEBACKUP2_E_NO_COMMON_VERSION'
now  I have no idea about it.

nikias, 2015-09-28:
Can you please compile libimobiledevice with debug output and run idevicebackup with -d option? My guess is either your libimobiledevice or any depedant library is tool old. Might even be libplist.

RonnieLLL, 2015-09-29:
I try to run idevicebackup with -d,  my libimobiledevice is the latest version, and I had update the libplist too. 
here is log-->
Backup directory is "C:\Users\Desktop\testbackup"
usbmuxd_connect: Reading connect result...
usbmuxd_connect: Connect success!
device udid: 5eee5e90e10ade52d5d7690fe5bf63aff60f5cd8
called
sending 287 bytes
sending 4 bytes
sending 287 bytes
sent 287 bytes
initial read=4
297 bytes following
received 297 bytes
success with type com.apple.mobile.lockdown
initialized config_dir to C:\ProgramData\Apple\Lockdown
sending 4840 bytes
sending 4 bytes
sending 4840 bytes
sent 4840 bytes
initial read=4
239 bytes following
received 239 bytes
ValidatePair success
internal pairing mode
using 30411473-7742861131734610232 as SystemBUID
sending 424 bytes
sending 4 bytes
sending 424 bytes
sent 424 bytes
initial read=4
354 bytes following
received 354 bytes
Session startup OK
SessionID: 11FA756E-71F7-4231-8BF5-7A7C70D61056
Enable SSL Session: true
SSL mode enabled, cipher: AES256-SHA
sending 364 bytes
sending 4 bytes
SSL_write 4, sent 4
sending 364 bytes
SSL_write 364, sent 364
sent 364 bytes
SSL_read 4, received 4
initial read=4
356 bytes following
SSL_read 356, received 356
received 356 bytes
usbmuxd_connect: Reading connect result...
usbmuxd_connect: Connect success!
Locked
Unlocked
starting callback.
Locked
Locked
sending 323 bytes
sending 4 bytes
sending 323 bytes
sent 323 bytes
Unlocked
Locked
NotificationProxy: no notification received!
Unlocked
sending 324 bytes
sending 4 bytes
sending 324 bytes
sent 324 bytes
Unlocked
Locked
sending 323 bytes
sending 4 bytes
sending 323 bytes
sent 323 bytes
Unlocked
Locked
sending 320 bytes
sending 4 bytes
sending 320 bytes
sent 320 bytes
Unlocked
sending 342 bytes
sending 4 bytes
SSL_write 4, sent 4
sending 342 bytes
SSL_write 342, sent 342
sent 342 bytes
SSL_read 4, received 4
initial read=4
334 bytes following
SSL_read 334, received 334
received 334 bytes
usbmuxd_connect: Reading connect result...
usbmuxd_connect: Connect success!
Adding escrow bag to StartService for com.apple.mobilebackup2
sending 437 bytes
sending 4 bytes
SSL_write 4, sent 4
sending 437 bytes
SSL_write 437, sent 437
sent 437 bytes
SSL_read 4, received 4
initial read=4
344 bytes following
SSL_read 344, received 344
received 344 bytes
Started "com.apple.mobilebackup2" service on port 49569.
usbmuxd_connect: Reading connect result...
usbmuxd_connect: Connect success!
Locked
initial read=4
80 bytes following
received 80 bytes
sending 91 bytes
sending 4 bytes
sending 91 bytes
sent 91 bytes
initial read=4
67 bytes following
received 67 bytes
sending 146 bytes
sending 4 bytes
sending 146 bytes
sent 146 bytes
NotificationProxy: no notification received!
Unlocked
initial read=4
176 bytes following
received 176 bytes
Locked
Could not perform backup protocol version exchange, error code -6

nikias, 2015-09-29:
You are not using the latest libimobiledevice. The debug output shows that it must be really old.
Especially during the version exchange you should get this error message: https://github.com/libimobiledevice/libimobiledevice/blob/master/src/device_link_service.c#L191

Please don't report issues when you're not using the latest version.

closed 2015-09-29 by nikias

N3gruVoda, 2021-02-09:
I have the same error even after downloading compiling and installing
 ideviceinstaller-1.1.1         
 idevicerestore-1.0.0         
 libimobiledevice-1.3.0         
 libplist-2.2.0        
 libusbmuxd-2.0.2
on ubuntu 20.04 and trying to backup an iphone 7 with IOS 14.4. :(

nikias, 2021-02-10:
@N3gruVoda libimobiledevice 1.3.0 is not the latest. You need code from git.

N3gruVoda, 2021-02-23:
@Nikias, it was the latest at the time when I tried. Git is where I had it
from.