libimobiledevice/libimobiledevice #239: Seem iOS 9 won't support archive application

mjkokid, 2015-10-05:
I tested archive some apps on IOS 9.0 -> 9.0.2 with both libimobiledevice and iTool. It didn't work! So do you have any comment or verify about this.

mexmer, 2015-10-07:
Archive function was removed from iOS ... this is due different packaging system, that will be used from iOS 9 onwards.
Search for Application thinning on google.

implementation details are here
https://developer.apple.com/library/prerelease/ios/documentation/IDEs/Conceptual/AppDistributionGuide/AppThinning/AppThinning.html

0x4d4746h, 2015-10-14:
@mexmer , is possible to get app's document & temp  and ... such as the functions of tool "iExplorer"

0x4d4746h, 2015-10-14:
[image]

mexmer, 2015-10-14:
view you have posted is trough utilizing house arrest service, which is not available for all application tho' .

archive function, that was available as part of installation proxy was removed.

0x4d4746h, 2015-10-14:
ok, thank you~