libimobiledevice/libimobiledevice #242: Through java code not able to read the data from terminal when,after enter the devicesyslog command

NVSateesh, 2015-10-08:
Through java code not able to read the data from terminal when,after enter the devicesyslog command
 
When we tried this command in Mac OS X (10.11) with iOS 6 device, Manully when we enter idevicesyslog in terminal, showing all the logs which are made by our app(any app, sits in the device.)..........the same command when we tried through java code not showing anything in the console.......   :( please give ur comments.......Thanks in advance :)

System.out.println("Case Started");
String[] str = { "/bin/bash", "-c", "idevicesyslog" }; 
		Process p = Runtime.getRuntime().exec(str);
	BufferedReader r = new BufferedReader(new InputStreamReader(p.getInputStream()));

		String line = "";
		String allLine = "";

		while ((line = r.readLine()) != null) {
		
			System.out.println(line);
}

System.out.println("Case Ended");

======== O/P =========

Case Started
Case Ended

if1else0, 2016-07-20:
Does it fixed?

abhilashkj, 2019-03-01:
got any solution?