libimobiledevice/libimobiledevice #243: ERROR: Could not connect to lockdownd, error code -2

Redskies, 2015-10-12:
Spend several hours trying to figure this out but with no luck. I reinstalled (libimobiledevice - 1.2.0) from the source (including libplist-1.12, ifuse-1.1.3 and libusbmuxd-1.1.0). I have no parental restrictions on the device and no screen lock. The device will not mount through linux and ifuse returns the same errors.

I am using CAINE linux (ubuntu 14.04) which might be the problem, i don't know. I double checked the lockdown file to ensure the permissions are correct (as you can see below.) I'm pretty much at a dead end.

Redskies:~$ idevice_id -l
da27a7e9b8150f5822c8d2cbeeb58f73baab1268

Redskies:~$ ideviceinfo -s
BasebandCertId: 3554301762
BasebandKeyHashInformation: 
AKeyStatus: 2
SKeyHash: 7MQEUyvzG4gjjZc7KsNNAVTS8g4=
SKeyStatus: 0
BasebandSerialNumber: M4KIHQ==
BasebandVersion: 6.00.00
BoardId: 10
BuildVersion: 13A344
ChipID: 35152
DeviceClass: iPhone
DeviceColor: #3b3b3c
DeviceName: iPhone
DieID: 1522290294251032272
HardwareModel: N48AP
PartitionType: 
ProductName: iPhone OS
ProductType: iPhone5,3
ProductVersion: 9.0
ProductionSOC: true
ProtocolVersion: 2
TelephonyCapability: true
UniqueChipID: 3787524586761
UniqueDeviceID: da27a7e9b8150f5822c8d2cbeeb58f73baab1268
WiFiAddress: 6c:94:f8:2a:80:81

Redskies:~$ ideviceinfo
ERROR: Could not connect to lockdownd, error code -2

Redskies:~$ idevicepair pair
ERROR: Please accept the trust dialog on the screen of device da27a7e9b8150f5822c8d2cbeeb58f73baab1268, then attempt to pair again.

Redskies:~$ idevicepair pair
SUCCESS: Paired with device da27a7e9b8150f5822c8d2cbeeb58f73baab1268

Redskies:~$ idevicepair validate
ERROR: Device da27a7e9b8150f5822c8d2cbeeb58f73baab1268 returned unhandled error code -2

Redskies:~$ ls -l /lib/udev/rules.d/
total 604
-rwxrwxrwx 1 root Redskies 611 Oct 21 2013 85-usbmuxd.rules

Redskies:~$ sudo chmod -R 0777 /var/lib/lockdown/

Redskies:~$ ls -l /var/lib/lockdown
total 12
-rwxrwxrwx 1 root root 7821 Oct 12 10:27 da27a7e9b8150f5822c8d2cbeeb58f73baab1268.plist
-rwxrwxrwx 1 root root 266 Oct 8 16:13 SystemConfiguration.plist

Redskies:/var/lib$ ls -l
total 316
drwxrwxrwx 2 root root 4096 Oct 12 10:27 lockdown

Redskies:/media$ ifuse /media/iphone
Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.

EDIT: when trying to run the system log

Redskies:~$ sudo idevicesyslog
ERROR: Could not start service com.apple.syslog_relay.
Could not start logger for udid da27a7e9b8150f5822c8d2cbeeb58f73baab1268
^C
Exiting...

Cinetik, 2015-10-14:
I have exactly the same issue right now. Same results on the same command.
Using an iPhone 5S 9.0.2 on Ubuntu 14.04 as well.

mexmer, 2015-10-14:
from my tests (i've done various test since ios 9 beta came out), syslog relay service is no longer accessible, when you try to start it, lockdown returns error InvalidService.

apple disabled various services and functions in iOS 9 to "harden" security.

nikias, 2015-10-14:
@Redskies your usbmuxd package is too old (not libusbmuxd!)
@mexmer syslog is still working in iOS 9.

mexmer, 2015-10-14:
@nikias you sure? when i try to start com.apple.syslog_relay i get "InvalidService" response.
Tested with and without passcode lock

nikias, 2015-10-14:
@mexmer absolutely sure. I've been using it all the time.

mexmer, 2015-10-14:
@nikias hmm, i will need to update ipad to oficial release then, i'm still using GM version from dev portal. might be, they reenabled it.

nikias, 2015-10-14:
@mexmer, it worked all the time, also with iOS 9 beta.

Cinetik, 2015-10-14:
Well I'm on the official iOS 9.0.2 OTA and it doesn't work for me... same issue as @Redskies

Redskies, 2015-10-14:
@nikias Will update and try again. And I will let you guys know if that works. 

I've since reinstalled Ubuntu 14.04 (just base distro) to see if that would work. I will update you guys soon.

mexmer, 2015-10-14:
@nikias i will test it after upgrade, and let you know.

Redskies, 2015-10-21:
Sorry for delay had an issue with old libplist library (1.10 was hiding in /usr/lib/x86_64). Compiled new libraries. 

Installed the new libimobiledevice and everything seems to work. BUT now i get this error

redskies:~$ idevice_id -l
ERROR: Unable to retrieve device list!
redskies:~$ idevicepair pair
No device found, is it plugged in?
redskies:~$ 

device is plugged and I select "trust" on the prompt. Is there anything I need to be doing to get the device to mount to ubuntu 14.04?

Redskies, 2015-10-21:
Success!

Issue was usbmuxd (for latest problem, see above post) 

manually activated usbmuxd and then paired the phone. ideviceinfo works

ran idevicebackup2 and it successfully backed up the phone. 

Thanks for all the help!

mexmer, 2015-10-26:
@nikias idk what was problem with syslog then. anyway, after upgrade of OS it started working, works also with latest dev beta of 9.1

hinxx, 2015-11-14:
This thread made access to my iPhone with IOS9 on Linux Mint 17 possible. Stock version of the libraries/tools failed to mount the drive. Using latest GIT code did the trick.
Thanks!!!

Redskies, 2015-11-17:
@hinxx  I"m glad it worked out for you!

jblanche, 2015-12-09:
I'm having the exact same error after compiling the latest versions of the tools, can it be because my device is running iOS 9.2 ? 

```
[Downloads]> usbmuxd --version
usbmuxd 1.1.1
[Downloads]> head -n1 libplist/NEWS 
Version 1.12
[Downloads]> head -n1 libusbmuxd/NEWS 
Version 1.0.10
[Downloads]> head -n1 libimobiledevice-1.2.0/NEWS 
Version 1.2.0
[Downloads]> idevice_id -l
d843d107b1897c65adf...
[Downloads]> ideviceinfo -s
DeviceName: Jonathan's iPhone
ProductName: iPhone OS
ProductType: iPhone6,2
ProductVersion: 9.2
[Downloads]> ideviceinfo
ERROR: Could not connect to lockdownd, error code -2
[Downloads]> idevicepair pair
SUCCESS: Paired with device d843d107b1897c65adf...
[Downloads]> idevicepair validate
ERROR: Device d843d107b1897c65adf... returned unhandled error code -2
[Downloads]> ios_webkit_debug_proxy 
Listing devices on :9221
Could not connect to lockdownd. Exiting.: Operation now in progress
Unable to attach d843d107b1897c65adf... inspector
```

Redskies, 2015-12-09:
Do you have the latest version of usbmuxd? (not libusbmuxd) 

Is it enabled? Manually run usbmuxd before starting to verify it is running. Then repeat the process. 

Where are you putting your new libraries? Did you make sure you deleted the old ones. The program may be trying to use the old libplist in /usr/lib instead of the newly compiled one. Make sure you have only the latest libraries. 

if your not putting libraries in /usr/lib or /usr/local/lib, do you have pkg-config set up properly? 

Did you check /var/lib/lockdown and make sure you have the right permisisons?
Hope we can get you in the right direction.

jblanche, 2015-12-09:
Deleting the old libraries did the trick :+1: 

Looks like it's working now. 

Thanks.

anand853, 2016-11-14:
even i have the same problem . can you please let me know the solution 
iOS (10.0.2) and OSX (10.11.6)