libimobiledevice/libimobiledevice #244: Add Support of application data folder back, please

Yanpas, 2015-10-13:
Since iOS 8, iOS has restructured application directory into three independent:
* Containers/Bundle
* Containers/Data
* Containers/Shared

Now I have no possibility to view Data directory and drop data files there. iOS 8.1, libimobiledevice 1.2

nikias, 2015-10-13:
This is something we're not in control of. Apple changed it. No way around it.

closed 2015-10-13 by nikias

Yanpas, 2015-10-13:
@nikias you just need to write some pieces of code. Why did you change status to "WontFix"? I still can drop some files to iOS apps via 3rd party programs on Windows. And there is still way to get appnames. Why it is impossible on linux?

mexmer, 2015-10-13:
it's not about code, since iOS 8.3 you don't have access to data directory, unless application has FileSharing feature enabled, and even then, you can only access documents folder and nothing else.

Yanpas, 2015-10-13:
@mexmer Documents folder is the most important. I was used to drop videos to player. Is it now impossible to implement mounting all "Documents" folders of all apps that allow to do it? In iOS 7 exactly this was possible

mexmer, 2015-10-13:
no, you can only access documents folder of application, that has document sharing feature enabled, like i said.

such applications shows in itunes under document sharing section.

if you enumerate apps and their properties (trough list application function of house arrest), you can check for value of "FileSharingUI" attribute, if it's set to true, then you can access documents folder, if it's set to false, it's not possible.

Yanpas, 2015-10-13:
@mexmer sounds fine! Exactly what most of us need and what we have lost since iOS 8. If it is possible, then it will be good enhancement. Can you reopen issue and mark it as a bug (cause it was possible before)?

nikias, 2015-10-13:
Still, this is not an issue with libimobiledevice. The API still works. You just use house arrest's VendContainer command on a supported app. You can use ```ifuse --container com.supported.app /mountpoint``` to get easy access to an apps data directory.

mexmer, 2015-10-13:
as @nikias there is no need to change in libimobile, since there is no change in protocol, you just need to change, how you actually use library.

also thanks to @nikias for example, i have don't ifuse, i have own scripts on top of libimobiledevice.

Yanpas, 2015-10-13:
Thanks, I've got the idea. I get error 
```
$ ifuse --container com.supported.app t/
Failed to start AFC service 'com.apple.mobile.house_arrest' on the device.
```
Do I need to report a new bug to ifuse, or there is a problem in my device? (iPhone 5, jailbreak, iOS 8.1)
PS I ve tried to connect via ssh to my iPhone and start this service from the device, but all to no avail `# launchctl load com.apple.mobile.house_arrest`

nikias, 2015-10-13:
@Yanpas you do understand that ```com.supported.app``` is just an example? You need to put the appid of the file sharing supported app there.

Yanpas, 2015-10-13:
@nikias Yes, I understand that that was an example, I've just cope-pasted your comment.
iOS7 jb iPad : `ifuse --documents org.videolan.vlc-ios t/` - mounts fine
iOS 8.1 iPhone: `ifuse --documents org.videolan.vlc-ios t/` - outputs error ***Failed to start AFC service 'com.apple.mobile.house_arrest' on the device.***

mexmer, 2015-10-14:
you sure, that vlc supports document sharing?

on ios 7 you could start housearrest for any application regardles it had documment sharing or not, on ios 8, only applications that support document sharing can be housearrested

for example try adobe acrobat reader, that app supports document sharing, so you can save pdf into it.

nikias, 2015-10-14:
For iOS 8 and later you need to use ```--container```.

Yanpas, 2015-10-14:
ideviceinstaller is broken on ios8 too. (I've installed adobe reader both on ipad and iphone). Here are commands with iOS8 device connected
```
$ ifuse --container com.adobe.Adobe-Reader t/
Failed to start AFC service 'com.apple.mobile.house_arrest' on the device.

$ ideviceinstaller -l |grep adobe
Could not start com.apple.mobile.installation_proxy!
```

zhuizhuhaomeng, 2015-12-14:
i got error  installation lookup failed
@Yanpas Yanpas

FerencYAO, 2015-12-16:
too 傻

mexmer, 2018-05-14:
you can't anymore, apple restricted it. you can only access shared documents for applications, that have document sharing enabled, and you can do that using ifuse command, iphone must be unlocked, when mounting using ifuse.

ifuse --documents <appid> <mountpoint>