libimobiledevice/libimobiledevice #248: Could not perform backup protocol version exchange, error code -6

RonnieLLL, 2015-10-19:
now i use  libimobiledevice(1.2.1) to backup mydevice.
the CMD: -u 5eee5e90e10ade52d5d7690fe5bf63aff60f5cd8 backup --full "C:\Users\Desktop\testbackup"

but It's failed, 
the error is:Could not perform backup protocol version exchange, error code -6
My Device: iPhone 5s 8.1.2

So I debug it and some info like down:
In function "device_link_service_version_exchange(..."
....
plist_t maj = plist_array_get_item(array, 1);
	plist_t min = plist_array_get_item(array, 2);
	uint64_t vmajor = 0;
	uint64_t vminor = 0;
	if (maj) {
		plist_get_uint_val(maj, &vmajor);
	}
	if (min) {
		plist_get_uint_val(min, &vminor);  /**************Here I can get the vmajor = 300, vminor = 0 *****************/
	}
	if (vmajor > version_major) {
		debug_info("Version mismatch: device=(%lld,%lld) > expected=(%lld,%lld)", vmajor, vminor, version_major, version_minor);
		err = DEVICE_LINK_SERVICE_E_BAD_VERSION;
		goto leave;
	} else if ((vmajor == version_major) && (vminor > version_minor)) {
		debug_info("WARNING: Version mismatch: device=(%lld,%lld) > expected=(%lld,%lld)", vmajor, vminor, version_major, version_minor);
		err = DEVICE_LINK_SERVICE_E_BAD_VERSION;
		goto leave;
	}
	plist_free(array);

	/* version is ok, send reply */
	array = plist_new_array();
	plist_array_append_item(array, plist_new_string("DLMessageVersionExchange"));
	plist_array_append_item(array, plist_new_string("DLVersionsOk"));
	plist_array_append_item(array, plist_new_uint(version_major));

        /******************************Here the array is****************************** 
             <?xml version="1.0" encoding="UTF-8"?>                        <!DOCTYPEplistPUBLIC"//Apple//DTDPLIST1.0//EN""http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
	<string>DLMessageVersionExchange</string>
	<string>DLVersionsOk</string>
	<integer>300</integer>
</array>
</plist>
        ***********************************************************************/

	if (property_list_service_send_binary_plist(client->parent, array) != PROPERTY_LIST_SERVICE_E_SUCCESS) {
		debug_info("Error when sending DLVersionsOk");
		err = DEVICE_LINK_SERVICE_E_MUX_ERROR;
		goto leave;
	}
	plist_free(array);
...

Now everything is OK,

But In Function "mobilebackup2_version_exchange(..."
......
int i;

	if (!client || !client->parent)
		return MOBILEBACKUP2_E_INVALID_ARG;

	plist_t dict = plist_new_dict();
	plist_t array = plist_new_array();
	for (i = 0; i < count; i++) {
		plist_array_append_item(array, plist_new_real(local_versions[i]));
	}

	plist_dict_set_item(dict, "SupportedProtocolVersions", array);

	mobilebackup2_error_t err = mobilebackup2_send_message(client, "Hello", dict);
/*********************Here dict is*************************
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>SupportedProtocolVersions</key>
	<array>
		<real>2.000000</real>
		<real>2.100000</real>
	</array>
</dict>
</plist>
**********************************************/

	plist_free(dict);

	if (err != MOBILEBACKUP2_E_SUCCESS)
		goto leave;

	char* Txml = NULL;
	uint32_t xmllength = 0;
	plist_to_xml(array, &Txml, &xmllength);

	dict = NULL;
	err = internal_mobilebackup2_receive_message(client, "Response", &dict);
	if (err != MOBILEBACKUP2_E_SUCCESS)
		goto leave;

	/* check if we received an error */
	plist_t node = plist_dict_get_item(dict, "ErrorCode");
	if (!node || (plist_get_node_type(node) != PLIST_UINT))
	{
		err = MOBILEBACKUP2_E_PLIST_ERROR;
		goto leave;
	}

/********************And Here is *****************************
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>ErrorCode</key>
	<integer>1</integer>
	<key>ErrorDescription</key>
	<string>No common supported protocol version</string>
	<key>MessageName</key>
	<string>Response</string>
</dict>
</plist>
*************************************************/

	uint64_t val = 0;
	plist_get_uint_val(node, &val);
	if (val != 0) {
		if (val == 1) {
			err = MOBILEBACKUP2_E_NO_COMMON_VERSION;
		} else {
			err = MOBILEBACKUP2_E_REPLY_NOT_OK;
		}
		goto leave;
	}
......

So , now I have no idea about this wrong, can you give some help for me?  many thanks!

RonnieLLL, 2015-10-19:
I compile and use the libimobiledevice(1.2.1) on Win7

RonnieLLL, 2015-10-23:
I have solved this problem. Because I hadn't configure the config.h when I compile the library on windows. i will close this issue as soon as! sorry about it.

closed 2015-10-23 by RonnieLLL

azerg, 2017-02-24:
@RonnieLLL, could you share any info how did you fix it 2 years ago :clown_face: ?

1714146, 2017-03-08:
@RonnieLLL Could u share any info how did u fix it 2 year ago? pls help me!

qmfrederik, 2017-08-18:
For those building from libimobiledevice-win32, this was caused by an issue in in the Visual C++ port of libplist. https://github.com/libimobiledevice-win32/libplist/commit/40aaaaae633f52e7b858c743162dc26720662211 fixes it. You can build from the libimobiledevice-win32 sources or download 1.2.1-r196 from http://docs.quamotion.mobi/en/latest/imobiledevice/download.html