libimobiledevice/libimobiledevice #249: Is this under the GPL or the LGPL?

kozross, 2015-10-20:
It's a bit unclear - you have the text of both licenses in your source tree, but you don't indicate which of these you're under.

nikias, 2015-10-20:
The library is LGPL while the tools are GPL. Each source file has a license header that states under which license it is.

kozross, 2015-10-20:
Thanks for clearing that up. It'd be nice to mention it in your README though.

nikias, 2015-10-21:
Also, citing from @saurik's comment here https://github.com/libimobiledevice/libplist/issues/48 :
"If you are releasing your program under the LGPL, you should also include the text version of the LGPL, usually in a file called COPYING.LESSER. Please note that, since the LGPL is a set of additional permissions on top of the GPL, it's important to include both licenses so users have all the materials they need to understand their rights."

closed 2016-05-12 by nikias