libimobiledevice/libimobiledevice #252: Unable to request source (Permission Denied -6)

hubertwang, 2015-11-04:
Hi, 

I am trying to retrieve system configuration with file_relay functions, but I always get FILE_RELAY_E_PERMISSION_DENIED error. Here is my codes (Objective-C) :

```
        const char *cLabel = "configuration";
        lockdownd_client_t lclient = NULL;

        if (lockdownd_client_new_with_handshake(device, &lclient, cLabel) == LOCKDOWN_E_SUCCESS)
            return;

        const char *serviceName = FILE_RELAY_SERVICE_NAME;
        lockdownd_service_descriptor_t service = NULL;

        if (lockdownd_start_service(lclient, serviceName, &service) != LOCKDOWN_E_SUCCESS)
            return;

        file_relay_client_t flClient = NULL;
        if (file_relay_client_new(device, service, &flClient) != FILE_RELAY_E_SUCCESS)
            return;

        const char *sources[] = {"SystemConfiguration", NULL};
        idevice_connection_t connection = NULL;
  
        if (file_relay_request_sources(rlClient, sources, &connection) == FILE_RELAY_E_SUCCESS) {

            char *data = NULL;
            uint32_t len = 1024;
            uint32_t recv_bytes = 0;
            
            NSMutableData *buffer = [NSMutableData data];
            
            while(idevice_connection_receive(connection, data, len, &recv_bytes)) {
                NSLog(@"len: %d, recv_bytes: %d", len, recv_bytes);
                [buffer appendBytes:data length:len];
            }
        }
```
Note that here is part of my code, so may lack something, but I am sure that creation of Lockdownd_Client, Lockdownd_Service_Descriptor, File_Relay_Client is OK. 
I am not sure if I miss something. Anyone know how to use the function?

mexmer, 2015-11-08:
filerelay service is not available since ios 8.
so if you trying to obtain filerelay files on ios 8 and newer, it will fail.

hubertwang, 2015-11-10:
@mexmer oh...that's pretty sad to hear... Thanks for your response.

FunkyM, 2015-11-10:
Not fully correct, the service is still available as you have successfully started it but there are added security measures to prevent it from being used easily.
Without the proper setup the service will return the permission denied error.

hubertwang, 2015-11-11:
@FunkyM  Yes, I successfully started the service. But I don't know what's the proper way to setup the service. Do you have any idea?

mexmer, 2015-11-14:
@FunkyM that's interesting, do you know if there any any other services, that are not denied per se, but have hardened security?

nikias, 2017-05-07:
AFAIK a configuration profile is required to be installed for this service to work. Anyway, this is not a bug in libimobiledevice.

closed 2017-05-07 by nikias