libimobiledevice/libimobiledevice #253: OS X EI Capitan Build Error

HDelbert, 2015-11-04:
>   CCLD     libimobiledevice.la
Undefined symbols for architecture x86_64:
  "_ERR_remove_thread_state", referenced from:
      _internal_idevice_deinit in idevice.o
      _idevice_connection_enable_ssl in idevice.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libimobiledevice.la] Error 1
make: *** [install-recursive] Error 1

Error occurs on link step.

deadlyfingers, 2015-11-04:
Also when doing `make` on *OS X 10.11* I now get **openssl** error
```fatal error: 'openssl/ssl.h' file not found```

HDelbert, 2015-11-05:
@deadlyfingers  
You can install openssl on *OS X 10.11* , like `brew install openssl`. The header files will be created at `/usr/local/Cellar/openssl/include/openssl` directory. You can make a link at `/usr/lib/include` directory, use this command `sudo ln -s /usr/local/Cellar/openssl/include/openssl /usr/lib/include/openssl`. This directories I not remember clear, so you can use this command `find /usr/ -name "ssl.h"` to find the directory.

HDelbert, 2015-11-05:
This problem is resolved. My openssl version was error. I link the `/usr/local/Cellar/openssl/1.0.2/lib/*` to `/usr/local/lib/`, then build OK.This problem is resolved. My openssl version was error. I link the `/usr/local/Cellar/openssl/1.0.2/lib/*` to `/usr/local/lib/`, then build OK.

closed 2015-11-05 by HDelbert

FunkyM, 2015-11-05:
Further note, if you have OpenSSL installed at unusual locations you can use the `openssl_CFLAGS` and `openssl_LIBS` environment variables to fix the build. No weired fiddling needed...

Check the configure output for more information:
```
$ ./configure --help
```

jflow, 2016-06-23:
I have used the openssl_CFLAGS and openssl_LIBS env variables but still get the same issues, have also done the symbolic link with the same results. I get the same results from fresh clones of the libimobiledevice repo. I have successfully build and installed libplist and libusbmuxd.

crummy, 2016-07-11:
I see the same error message building on OSX 10.11.5:

```
Undefined symbols for architecture x86_64:
  "_ERR_remove_thread_state", referenced from:
      _internal_idevice_deinit in idevice.o
      _idevice_connection_enable_ssl in idevice.o
ld: symbol(s) not found for architecture x86_64
```

I got the `fatal error: 'openssl/ssl.h' file not found` error also mentioned above, which I resolved with `brew link openssl --force`.

What values should I use for `openssl_CFLAGS` and `openssl_LIBS` variables to successfully build?

brunoro, 2016-08-03:
got it running by manually setting `CFLAGS` and `LDFLAGS` manually:
`CFLAGS=-I/usr/local/opt/openssl/include LDFLAGS=-L/usr/local/opt/openssl/lib ./configure`