libimobiledevice/libimobiledevice #254: SSLv3_method issue on compile

M1ke, 2015-11-09:
/src/.libs/libimobiledevice.so: undefined reference to `SSLv3_method'

After a successful `./configure` step the `make` process fails with the above error when compiling on Ubuntu 14.04.3 LTS

I am unsure what versions or files would help to diagnose this. I have updated cURL to the latest version and looked at the `openssl_CFLAGS` option in configure but am unsure what to set it to.

I assume the bug relates to SSLv3 now being known to be insecure and therefore deprecated in other packages.

FunkyM, 2015-11-10:
Could you retry with the following patch?
```
--- libimobiledevice-1.2.0.orig/src/idevice.c
+++ libimobiledevice-1.2.0/src/idevice.c
@@ -678,7 +678,7 @@ LIBIMOBILEDEVICE_API idevice_error_t ide
 	}
 	BIO_set_fd(ssl_bio, (int)(long)connection->data, BIO_NOCLOSE);
 
-	SSL_CTX *ssl_ctx = SSL_CTX_new(SSLv3_method());
+	SSL_CTX *ssl_ctx = SSL_CTX_new(SSLv23_method());
 	if (ssl_ctx == NULL) {
 		debug_info("ERROR: Could not create SSL context.");
 		BIO_free(ssl_bio);
```

Thanks.

M1ke, 2015-11-12:
That worked to compile `libimobiledevice-1.2.0`. I've now also succesfully compiled `ios-webkit-debug-proxy-master` which depends on it. Thanks for the fix!

xros, 2015-12-02:
Thanks. I had the same problem. And it works.

marijnh, 2016-02-26:
(I just ran into this too, and applying the change helped. Maybe commit it?)

nikias, 2016-06-30:
Should be resolved now with latest code.

closed 2016-06-30 by nikias