libimobiledevice/libimobiledevice #260: when I build the project, I got following error. Please help to resolve.

qqtester, 2015-11-26:
when I run "make", I got this:
samynyins-iMac:libimobiledevice samynyin$ make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in common
  CC       socket.lo
  CC       thread.lo
  CC       debug.lo
  CC       userpref.lo
  CC       utils.lo
  CCLD     libinternalcommon.la
Making all in src
  CC       idevice.lo
  CC       service.lo
  CC       property_list_service.lo
  CC       device_link_service.lo
  CC       lockdown.lo
  CC       afc.lo
  CC       file_relay.lo
  CC       notification_proxy.lo
  CC       installation_proxy.lo
  CC       sbservices.lo
  CC       mobile_image_mounter.lo
  CC       screenshotr.lo
  CC       mobilesync.lo
  CC       mobilebackup.lo
  CC       house_arrest.lo
  CC       mobilebackup2.lo
  CC       misagent.lo
  CC       restore.lo
  CC       diagnostics_relay.lo
  CC       heartbeat.lo
  CC       debugserver.lo
  CC       webinspector.lo
  CC       syslog_relay.lo
  CCLD     libimobiledevice.la
Undefined symbols for architecture x86_64:
  "_ERR_remove_thread_state", referenced from:
      _internal_idevice_deinit in idevice.o
      _idevice_connection_enable_ssl in idevice.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

qqtester, 2015-11-26:
I used Xcode7, OSX10.11.1

qqtester, 2016-04-01:
I have fix it. closed.

ledinhphuong, 2016-04-22:
@qqtester I face the same issue with you. Could you please share how can you fix it?

qqtester, 2016-04-25:
@ledinhphuong ,check your openssl version, it must > 1.0.0, according to the code. you can check it in idevice.c.

icrazyfox72, 2016-06-05:
@qqtester 
The openssl version is 1.0.2h_1 which I use.But also get this issue...
At first,I get the fatal error :
[image]
and I copy the openssl dir from the '/usr/local/Cellar/openssl' to '/Library/Developer/CommandLineTools/usr/include',then face the issue like you.

qqtester, 2016-06-06:
@icrazyfox72 , 
$openssl
-version
check your openssl version which being used. maybe you install a new version openssl ,but still using old version in your os x.
if so,
rm /usr/bin/openssl
ln –s /usr/local/Cellar/openssl/x.x.x/bin/openssl /usr/local/bin
Hope this works.

icrazyfox72, 2016-06-07:
@qqtester 
Thank you.I fixed it ,follow your operation.

closed 2016-10-10 by qqtester

inspectionsbybob, 2016-10-10:
Has anyone built a full, complete, build script that once put in a 'temp' folder and given privileges goes and gets EVERYTHING needed to rebuild libimobiledevice and update/replace the current version on a particular platform?

I know that Hayden Young
was trying to build a script at
https://github.com/mirukaiyt/libimobile-installer
 
but it took forever to run, seemed to touch EVERYTHING (and I mean everything), and -still- gave me a mount error when i plugged in my ipad to a Linux mint 17.3 system.....