libimobiledevice/libimobiledevice #263: Fetching IOKit tree via idevicediagnostics ioreg failing on iOS 9.1

lyonanderson, 2015-12-01:
Trying any of the planes returns an empty tree:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>IORegistry</key>
	<dict>
	</dict>
</dict>
</plist>
```

On the iOS device (tried iPad Air 2 and  iPhone 6S+) there is an error in the console:

```
mobile_diagnostics_relay[301] <Error>: create_ioreg_dict: IORegistryEntryGetLocationInPlane has failed with error -536870160.
```

Kanong017, 2016-01-24:
The function used in the original diagnostics_relay_query_ioregistry_plane is wrong.
Should use diagnostics_relay_query_ioregistry_entry
Class as""

Marvinh5, 2016-01-26:
@Kanong017  what do you mean with Class as ""

Kanong017, 2016-01-26:
diagnostics_relay_query_ioregistry_entry
There are four parameters
1.diagnostics client
2.class  Here is null
3.name  Similar to IOCPU
4.return node

Then plist_to_xml(node) can

lyonanderson, 2016-01-26:
@Kanong017  this seems to work well. However, you do need to know a load of key names as there does not seem to be way of dumping the whole plain.

Kanong017, 2016-01-26:
Anyway, I use this method to get the battery details can be

Marvinh5, 2016-01-26:
@Kanong017  what name do you pass to get the battery details?

lyonanderson, 2016-01-26:
You can try AppleARMPMUCharger

Marvinh5, 2016-01-26:
Tu that worked , love you guys

Kanong017, 2016-01-27:
@lyonanderson Do you know how to read the type of NAND, such as or MLR TLC?

lyonanderson, 2016-01-27:
@Kanong017  you could try ASPStorage

Kanong017, 2016-01-27:
@lyonanderson Thanks!

daniel-bluesea, 2017-10-01:
the "AppleARMPMUCharger" & "ASPStorage" keys are working fine but still "IODeviceTree" returns empty..

does it have a "new" name? where can i find the list of "enums" i can pass to the diagnostics_relay_query_ioregistry_entry function?

thanks in advance!

truonggiang0710, 2017-11-02:
Yes ! I'm in the same issue ! I don't know how to fix it !