libimobiledevice/libimobiledevice #272: idevicedebug logging is rediculously slow

chebz, 2015-12-17:
If you try capture logs with idevicedebug they are incredibly slow. This is game breaking for me.

brunoro, 2016-08-19:
I played around with this and the main culprit is a `sleep(1)` call in the main loop that adds a 1 sec wait before fetching data from the device. It's possible to drastically reduce this delay time by moving the sleep call to the beginning of the loop. A smaller delay in the end of the loop would cause a crash.

This commit on my fork does exactly that: https://github.com/brunoro/libimobiledevice/commit/2727fd508145018ce532bf525bd0b5383ce6b1f3
Is there any interest on a pull request? Also, I haven't found any documentation regarding the 1 sec sleep call, so I'm not sure if my solution has any pitfalls.

dolphineye, 2018-11-16:
Hi, we have the same issue here. Any update on this topic? Regards!

anton-kl, 2020-11-03:
One workaround is to run the app with `OS_ACTIVITY_DT_MODE=enable` like this:
`idevicedebug -e OS_ACTIVITY_DT_MODE=enable run com.my.app`
This will ensure NSLog, CFLog and os_log messages get mirrored to stderr (see https://github.com/ios-control/ios-deploy/blob/master/src/scripts/lldb.py#L55)
Then you can use `idevicesyslog` to get logs from device, which will include logs from your app - in real time.