libimobiledevice/libimobiledevice #275: com.apple.fmip not working

0xFireball, 2016-01-04:
Looking at numerous open-source projects and jailbreaking tools such as TaiG and Pangu, *Find my iPhone* can be detected by querying the domain 'com.apple.fmip'

However, ideviceinfo continues to return an empty result, yet other programs are still able to find this information.

I have confirmed on Windows, OSX, Ubuntu, and Debian.

nikias, 2016-01-04:
Have you tried:
```
ideviceinfo -q com.apple.fmip -k IsAssociated
```

0xFireball, 2016-01-04:
That works; 
but how come it is not listed when querying com.apple.fmip

closed 2016-01-04 by 0xFireball

nikias, 2016-01-04:
There a several domains that behave like this.

Wombo1010, 2017-02-16:
Hi,  why am getting this error? is com.apple.fmip no longer works?  Please help.
ideviceinfo -q com.apple.fmip -k IsAssociated WARNING: Sending query with unknown domain "com.apple.fmip". false

nemecence90, 2017-02-17:
@Wombo1010 You are already getting a response.
# ideviceinfo -q com.apple.fmip -k IsAssociated 
WARNING: Sending query with unknown domain "com.apple.fmip".
**false <- This is your FMIP response.**

Wombo1010, 2017-02-17:
Thank you Funky for your response.  However, the response is not correct, i have a device with fmip activated but the response is FALSE.  Should it be TRUE ?  is there another way to get the correct response?  thank you.

Wombo1010, 2017-02-17:
Thank you nemecence90.

FunkyM, 2017-02-17:
Works for me (iOS 10.3). Furthermore still there is no code issue to discuss here.
The retrieval of lockdown values works fine.
What value is returned is not in the project's scope but depends on your device and Apple's iOS version.
It would need proper investigation and we currently have no focus on this. :fog:

Wombo1010, 2017-02-17:
I tested on several units with iOS 10.2,  iphone 6,  5S and 6S.  All these have fmip lock on but all response with a FALSE.  I just wanted to know if am i missing something here.  Any help is appreciated.   
 "ideviceinfo -q com.apple.fmip -k IsAssociated"

Wombo1010, 2017-02-17:
It's working now.  Thank you

daniel-bluesea, 2018-02-05:
i want to add another scenario:
if somehow you managed to "Erase All Content and Settings" without entering your appleid credentials,
after reboot the device will return "false" to the "ideviceinfo -q com.apple.fmip -k IsAssociated" query, but! the device is still associated to iCloud

how can we identify this kind of scenarios?

daniel-bluesea, 2018-02-06:
also the ActivationInfo that is created and is sent to apple activation server during activation shows:
<key>FMiPAccountExists</key>
<false/>

but of course the server will ask for appleid credentials because the device has activation lock and its find my iPhone still on (you can check it on iCloud.com).

is it an Apple bug?

daniel-bluesea, 2018-02-13:
@Wombo1010 @nemecence90 ideviceinfo -q com.apple.fmip -k IsAssociated fails to return correct answer for erased devices in setup wizard stage, its returning "false" regardless of the real status.