libimobiledevice/libimobiledevice #276: ERROR: Could not connect to lockdownd, error code -5

divyum, 2016-01-08:
Hi,
None of the `libimobileidevice` commands work with this error. I am facing this error on ios 9. I tried to reboot the device but didn't help. 
As per the source -5 code is LOCKDOWN_E_SSL_ERROR.
I even tried to change time of device to my mac time but still facing the issue.

I am using libimobiledevice-1.2.0

mexmer, 2016-01-10:
this error comes from wrong pairing information.
you need to redo pairing with your phone first.

divyum, 2016-01-11:
Thanks. It helped :+1:

closed 2016-01-11 by divyum

templatetuners, 2016-08-28:
i'm getting this error "ERROR: Could not connect to lockdownd, error code -5" on iOS 9.2. Paired with the phone with idevicepair pair and got SUCCESS: Paired with device 76ac1f724a6bd1e80ba55509f9ff4faa099d398f but still error -5
using libi 1.2.0 latest version. any help?

templatetuners, 2016-09-01:
@VasilijSviridov you can try [https://mega.nz/#!V0dUAQqT!CBca69kGYpGQObiKuO1LU9ROqnXakThp5a0-8Y3Svfw](url) - works for me

VasilijSviridov, 2016-09-02:
Do you have similar for mac osx users ?

templatetuners, 2016-09-02:
[https://mega.nz/#!RhtUwLJB!pNyM7lLEStfiDdsl0CRhXMZnE6sTrgPD12mUqRZBnuU](url) New Updated LibiMobileDevice [iDeviceRestore] Compiled - OS X

VasilijSviridov, 2016-09-02:
Thank you so much will test it
Cheers

VasilijSviridov, 2016-09-02:
@templatetuners  could you tell steps how you compile libimobile* for using tls instead of ssl ?

burakbozyigit, 2017-01-06:
Hey there i am still getting this error on Mac OSX. Do you have any idea to solve this permanently?
I've installed package via npm.

nikias, 2017-01-06:
1.2.0 is not the latest, you need the latest code from git. lockdown error -5 means SSL error. You need to link against a working version of OpenSSL, the one from MacPorts works fine. Also, there is no need in using ```idevicepair``` because macOS stock usbmuxd will pair the device for you and the idevicetools should just work.

burakbozyigit, 2017-01-06:
Thanks for the answer @nikias ! 
I also apologize for being an impatient man as after my comment, I upgraded related package and get rid of the problem. 
Thanks again and sorry for impatient behaviour.

crummy, 2017-01-25:
I see this issue on Mac with iOS10 devices, after a `brew install libimobiledevice`. Fixed with `brew install --HEAD libimobiledevice` (you may have to unlink first).

TiagoGomesDesign, 2017-08-22:
Thanks @crummy! That work for me