libimobiledevice/libimobiledevice #281: Any domain or keys to check jailbroken status just like "com.apple.fmip" to check find my iphone?

jackczhang, 2016-02-16:
Is there any domains or keys to check if the device jailbroken?
Just like com.apple.fmip key IsAssociated to check "find my iphone"?

mexmer, 2016-02-16:
you can check if there is afc2 service running (but this is optional), you can also check if ssh port is open (either 22 or 2222), but that's also optional.

afaik none of jailbreaks offers some aditional service, and they rather try to be hidden.

also another test (for jailbreak) is to try to install selfsigned application. jailbreak is primarily for allowing installation of nonstore application (eg. disables signature validity check) - imo this one is most reliable.

closed 2016-02-26 by jackczhang

daniel-bluesea, 2018-02-13:
@mexmer thanks for your reply.
by selfsigned application do you mean signed with my developer signature?

mexmer, 2018-02-13:
yes, for example. jailbroken device will accept developer signed package even if it's not on list of testing devices.

daniel-bluesea, 2018-02-14:
cool, i'll check this out
thanks