libimobiledevice/libimobiledevice #283: Unable to build libimobieldevice with python bindings

parkerlreed, 2016-02-18:
Arch x86_64
libimobiledevice-git from AUR (Uses the sukimashita repo)

libplist IS installed with cython bindings. libimobiledevice configure seems to think it isn't.

No matter what I do to libimobiledevice's configure file I cannot seem build the bindings. cython2 is installed and worked for libplist.

````
[parker@e73 libimobiledevice]$ PYTHON=/usr/bin/python2 ./autogen.sh --prefix=/usr
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /usr/bin/automake line 3936.
configure.ac:31: installing './compile'
configure.ac:34: installing './config.guess'
configure.ac:34: installing './config.sub'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
common/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
[... 139 lines cut ...]
libplist-git /usr/include/plist/Date.h
libplist-git /usr/include/plist/Dictionary.h
libplist-git /usr/include/plist/Integer.h
libplist-git /usr/include/plist/Key.h
libplist-git /usr/include/plist/Node.h
libplist-git /usr/include/plist/Real.h
libplist-git /usr/include/plist/String.h
libplist-git /usr/include/plist/Structure.h
libplist-git /usr/include/plist/Uid.h
libplist-git /usr/include/plist/plist++.h
libplist-git /usr/include/plist/plist.h
libplist-git /usr/lib/
libplist-git /usr/lib/libplist++.so
libplist-git /usr/lib/libplist++.so.3
libplist-git /usr/lib/libplist++.so.3.0.0
libplist-git /usr/lib/libplist.so
libplist-git /usr/lib/libplist.so.3
libplist-git /usr/lib/libplist.so.3.0.0
libplist-git /usr/lib/pkgconfig/
libplist-git /usr/lib/pkgconfig/libplist++.pc
libplist-git /usr/lib/pkgconfig/libplist.pc
libplist-git /usr/lib/python2.7/
libplist-git /usr/lib/python2.7/site-packages/
libplist-git /usr/lib/python2.7/site-packages/plist.so
````

Tatsh, 2016-03-17:
`configure: WARNING: cannot find libplist Cython bindings. You should install your distribution specific libplist Cython bindings package.`

Your file listing is missing the pxd file.

```
$  qlist libplist | sort -u
/usr/bin/plistutil
/usr/bin/plutil
/usr/include/plist/Array.h
/usr/include/plist/Boolean.h
/usr/include/plist/Data.h
/usr/include/plist/Date.h
/usr/include/plist/Dictionary.h
/usr/include/plist/Integer.h
/usr/include/plist/Key.h
/usr/include/plist/Node.h
/usr/include/plist/plist.h
/usr/include/plist/plist++.h
/usr/include/plist/Real.h
/usr/include/plist/String.h
/usr/include/plist/Structure.h
/usr/include/plist/Uid.h
/usr/lib64/libplist.a
/usr/lib64/libplist++.a
/usr/lib64/libplist.so
/usr/lib64/libplist++.so
/usr/lib64/libplist.so.3
/usr/lib64/libplist++.so.3
/usr/lib64/libplist.so.3.0.0
/usr/lib64/libplist++.so.3.0.0
/usr/lib64/pkgconfig/libplist.pc
/usr/lib64/pkgconfig/libplist++.pc
```
**`/usr/lib64/python2.7/site-packages/libplist/includes/plist.pxd`**
```
/usr/lib64/python2.7/site-packages/plist.la
/usr/lib64/python2.7/site-packages/plist.so
```
**`/usr/lib64/python3.4/site-packages/libplist/includes/plist.pxd`**
```
/usr/lib64/python3.4/site-packages/plist.la
/usr/lib64/python3.4/site-packages/plist.so
/usr/share/doc/libplist-9999/AUTHORS.bz2
/usr/share/doc/libplist-9999/NEWS.bz2
/usr/share/doc/libplist-9999/README.bz2
```

aymenim, 2016-03-18:
@parkerlreed when libimobiledevice configure checks if cython binding of libplist exists it checks for
```
 $include_dir/plist/cython/
```
from configure:
```
if test "x$CYTHON" != "xfalse"; then
            { $as_echo "$as_me:${as_lineno-$LINENO}: checking for libplist Cython bindings" >&5
$as_echo_n "checking for libplist Cython bindings... " >&6; }
            CYTHON_PLIST_INCLUDE_DIR=$($PKG_CONFIG --variable=includedir libplist)/plist/cython
            if false; then
                CYTHON=false
                CYTHON_SUB=
                cython_python_bindings=no
                { $as_echo "$as_me:${as_lineno-$LINENO}: result: no" >&5
$as_echo "no" >&6; }
                { $as_echo "$as_me:${as_lineno-$LINENO}: WARNING: cannot find libplist Cython bindings. You should install your distribution specific libplist Cython bindings package." >&5
$as_echo "$as_me: WARNING: cannot find libplist Cython bindings. You should install your distribution specific libplist Cython bindings package." >&2;}
            else

                { $as_echo "$as_me:${as_lineno-$LINENO}: result: $CYTHON_PLIST_INCLUDE_DIR" >&5
$as_echo "$CYTHON_PLIST_INCLUDE_DIR" >&6; }
                CYTHON_SUB=cython
                cython_python_bindings=yes
            fi
else
            CYTHON_SUB=
            cython_python_bindings=no
fi
```  
an easy solution would be to create a cython dir inside the libplist include dir and copy the plist.pxd to there, so it would have this kind of structure
```
libplist/1.12/include
└── plist
    ├── Array.h
    ├── Boolean.h
    ├── Data.h
    ├── Date.h
    ├── Dictionary.h
    ├── Integer.h
    ├── Key.h
    ├── Node.h
    ├── Real.h
    ├── String.h
    ├── Structure.h
    ├── Uid.h
    ├── cython
    │   └── plist.pxd
    ├── plist++.h
    └── plist.h
``` 
this will fix it.

parkerlreed, 2016-03-18:
I think that worked but now I don't even remember what I was trying to do exactly. Thanks for the replies!

closed 2016-03-18 by parkerlreed

mattwomple, 2019-05-10:
In case anyone else stumbles upon this issue, on Ubuntu 18.04, I just needed to install the `python-plist` package.