libimobiledevice/libimobiledevice #284: Can only read a few files via ifuse: NSDescription="No file slots"

UnitedMarsupials, 2016-02-21:
Hello!

I am able to successfully mount my phone -- so far so good. I then kick off `rsync` to get the new photos (and videos) off the device:

    rsync -av /iphone/DCIM/*/* /m/iPhone/

and *this* part does not quite work -- only a few files are read before rsync begins reporting "Input/output error" and/or "Operation timed out". At the same time `idevicesyslog` floods my xterm with identical messages like:

    Feb 21 13:43:13 Misa afcd[6174] <Error>: AFCFileLine="1922" AFCFileName="server.c" AFCCode="-402636797" NSDescription="No file slots" AFCVersion="245"

I am using FreeBSD-10.2 with the currently-latest versions of all components listed on http://www.libimobiledevice.org/ :

- fusefs-ifuse-1.1.3_2
- usbmuxd-1.1.0
- libusbmuxd-1.0.10
- libplist-1.12
- libimobiledevice-1.2.0

Simply remounting (`SIGKILL`-ing ifuse and starting it anew) does not help, but unplugging the phone sometimes allows to get a few more files. Not always:

    root@... (1026) ifuse -o allow_other /iphone
    root@... (1027) rsync -av /iphone/DCIM/*/* /m/iPhone/
    sending incremental file list
    rsync: readlink_stat("/iphone/DCIM/100APPLE/IMG_0853.JPG") failed: Input/output error (5)
    rsync: link_stat "/iphone/DCIM/100APPLE/IMG_0855.JPG" failed: Input/output error (5)
    rsync: link_stat "/iphone/DCIM/100APPLE/IMG_0857.JPG" failed: Input/output error (5)
    ...

The "lightning" cable is by Amazon -- and the phone does not complain about it being "incompatible". I regularly use the USB-extension, into which I plug it, for getting files from flash-cards without a problem.

Any suggestions? Thank you!

eriknstr, 2017-01-14:
I have similar problems with my iPod touch and a computer running FreeBSD 11.0-RELEASE. usbmuxd 1.1.0, ifuse 1.1.3. My iPod touch is running iOS 9.3.5. Did you find any solution to this problem?

CarlMilnerErnest, 2018-12-06:
Identical problem to the original poster.

I'm running FreeBSD 11.2-STABLE, and have checked out and built the latest versions of libusbmuxd, libimobiledevice, usbmuxd, and ifuse from git.

ideviceinfo and other idevice info-showing programs all work perfectly.

ifuse will successfully mount the device, and I can list out the filesystem successfully below the mount point.  But attempts to read any files will start generating I/O errors after reading a few files.

The phone, cable, and underlying host USB setup all seem fine.  I can use gphotofs or gphoto2 CLI to read all the photos off the phone via MTP.  I have been using MTP with this same phone/cable/computer for a couple of years now, and it is always fast and reliable.

Please let me know what info I can provide to help track this down, or send any pointers to how I might help debug this issue.

Thanks.

CarlMilnerErnest, 2018-12-07:
Interestingly, when I tried running usbmuxd in debug mode (-v -f) and ran idevicesyslog before mounting with ifuse (to try and gather some debugging data), I stopped getting any I/O errors from the ifuse mount.  Transfers went very slowly (~3 hours to rsync 20ish gigs of data), but ultimately completed without error.

Totally wild, baseless guess, based in part on OP's original debug logs: maybe the host is sending commands to the iPhone too quickly, and overflowing some buffer; but when using the debug tools, this slows the host down enough to prevent this?

Still hoping one of the devs can weigh in with an opinion on this one.

If not, then probably my next step would be to try and add some rate-limiting to the ifuse code.

Thanks.

nikias, 2018-12-07:
From what I can see the error means that too many files are open.

nikias, 2018-12-09:
I checked the code in libafc.dylib and it seems like the maximum number of open files is 231 (checked on an iPhone 6) so it seems like it is not "closing" the files. However since it works when debugging maybe it just delays too long before closing files? Really weird...