libimobiledevice/libimobiledevice #292: how do i porting  iphone tethering to arm linux

alexwangf, 2016-03-13:
hi all,
     i want to porting libimobiledevice,libusbmuxd,usbmuxd to linux arm platform. i want follow below way,is it right.
      1. when system boot, ipheth (linux ) will be installed
      when iphone insert to the arm platform, 
       2. call usbmuxd  
       3. call idevicepair --validate(from libimobiledevice ),if return ok,goto step 4,if fail,call idevicepair    --pair
       4. call DHCP by the net interface USBXX(from ipheth)

mikesumsun, 2016-03-30:
any luck?

alexwangf, 2016-04-25:
fix it already

sancelot, 2019-11-18:
From usbmuxd readme page, it looks like usbmuxd does not seem useful to enable tethering.

superseani2, 2020-11-20:
I mean on my old Sw900w8 (galaxy s5) + root.

I believe it works. i got the "Checkm8" iresponse