libimobiledevice/libimobiledevice #294: Disable SSLv3 - lockdown error: -5

Qwertylex, 2016-03-16:
I have an iPad Mini (1st gen) on iOS 6.1.3, after upgrading Arch Linux, which upgraded packages: 
```
libimobiledevice (from 1.2.0-1 to 1.2.0-3) and openssl (from 1.0.2.f-1 to 1.0.2.g-3) 
(the Disable SSLv3 patch)
```
My iPad Mini stopped connecting to linux, leaving messages:

> usbmuxd: preflight_worker_handle_device_add: The stored pair record for device a6de54<redacted> is invalid. Removing.
> usbmuxd: preflight_worker_handle_device_add: ERROR StartSession failed on device a6de54<redacted>, lockdown error -5

I deleted /var/lib/lockdown and I reinstalled libimobiledevice and usbmuxd, same issue.

I downgraded libimobiledevice and openssl and everything worked normally.

hasnainbangash, 2016-04-10:
You are right sir. I had the same problem and I tried everything till I came across your post.
I was getting the following error:
$ ifuse iphone/ 
Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.

and also 

ERROR: Could not connect to lockdownd, error "code -5"

I am running arch linux and did the following:
I went to /var/cache/pacman/pkg and downgraded to libimobiledevice 1.2.0-1 and openssl 1.0.2.e-1.

I am currently running libimobiledevice 1.2.0-1, 1.0.2.e-1, libplist 1.12-4, usbmuxd 1.1.0-2, ifuse 1.1.3-3, ideviceinstaller-git 1.1.0.r12.g824bb9a-1, libideviceactivation 1.0.0-3, libusbmuxd 1.0.10-1, and libplist 1.12-4.

It is all working now. Thank you Qwertylex

mexmer, 2016-04-10:
elder version of use SSLv3, so if you use openssl without SSLv3, it will fail. Newer versions also support TLS
problem is not in libimobile itself, but in openssl library you use

Qwertylex, 2016-04-17:
Supposedly the old iOS device I have does not support TLS.

Anachron, 2016-04-17:
Subscribing, I also experience the same issue here.

mexmer, 2016-04-18:
@Qwertylex no they don't support TLS, that's why you need SSLv3

randomize, 2016-05-14:
As an alternative to use old iOS devices in Arch is building latest upstream libimobiledevice with `--disable-openssl` - in this case GnuTLS will be used as backend and it still has SSLv3 enabled in Arch repos at this moment. The Arch way will be to just write simple PKGBUILD that uses git source and passes that flag to configure -- here is example [PKGBUILD](https://gist.github.com/randomize/98f4e77a09f88f9a97adf843daf6d6bc) - quick and dirty, but worked for me, can be cleaned up to have pkgver support for example.

andrewmichaelsmith, 2016-05-20:
Just to add to this - right now I'm getting "lockdown error: -5" when trying to run `ideviceinfo`.

I tried @randomize's trick which appears to have switched my compile to GnuTLS:

```
Configuration for libimobiledevice 1.2.1:
-------------------------------------------

  Install prefix: .........: /usr
  Debug code ..............: no
  Python bindings .........: no
  SSL support backend .....: GnuTLS
```

But I still get the -5 error when trying `ideviceinfo`. (If it's relevant) I built usbmuxd off master.

webavant, 2016-05-24:
My comment will be off-topic, since I'm addressing a makepkg issue, not a libimobiledevice issue, but just to help the above commenter, @andrewmichaelsmith:

If you have already installed the libimobiledevice-git package from AUR, you will need to locally create the PKGBUILD file that @randomize suggested, then run `makepkg -if` within its directory in order to rebuild and force install the package.

mexmer, 2016-05-24:
@webavant there are 3 options, you pick, which suits you best
1) use gnutls (eg. --disable-openssl)
2) downgrade openssl to version with SSLv3 support (not recommended)
3) recompile openssl with SSLv3 support ... it was not removed from openssl yet, it's just disabled in default configuration, which is used by most distributions

webavant, 2016-05-24:
@mexmer thanks for clarifying! Should be helpful for anyone confused. I meant to answer @andrewmichaelsmith's question, although now I'm realizing my response wasn't even appropriate for his problem, although it might be helpful to anyone unfamiliar with makepkg. Feel free to delete.

nikias, 2016-06-15:
Can you check if it works with my latest commit? SSLv23_method() should be available from my understanding.

nikias, 2016-06-30:
Should be fixed with latest code from git master.

closed 2016-06-30 by nikias

VasilijSviridov, 2016-09-02:
how to use libimobiledevice on mac osx for accessing iphone device with ios10 ?
At the moment I am getting Lockdown -5 which is ssl issue 

Please help

pmnordkvist, 2016-09-05:
Gaah! This is killing me! I have compiled several times with latest master and --disable-openssl option. Always get the "GnuTLS error: Error in the pull function." message when running ideviceinfo or ifuse. paring and validating works as expected. 

I have also tried to downgrade openssl to "1.0.1e" and compile with openssl but that does not work either. Same result: "GnuTLS error: Error in the pull function.". Should I even get that message when using openssl? Have been testing 3 different devices on versions 9.3.5 and 9.3.2 and I'm running out of ideas. 

Any specific version of gnutls required for this to work? My best guess right now is bad gnutls version.

joecool1029, 2016-09-19:
@pmnordkvist https://github.com/libimobiledevice/libimobiledevice/issues/365#issuecomment-247911953 rebuild nettle and then gnutls and let me know how goes.

VasilijSviridov, 2016-09-19:
this worked for me https://github.com/appium/appium/issues/6597#issuecomment-247038942 =)

joecool1029, 2016-09-19:
@VasilijSviridov that is a totally unrelated issue on OSX. I am on Linux.

VasilijSviridov, 2016-09-20:
@joecool1029  thought this will help you : brew uninstall libimobiledevice && brew install --HEAD libimobiledevice

joecool1029, 2016-09-20:
@VasilijSviridov you thought wrong

nikias, 2016-11-04:
The pairing record generation with newer GnuTLS was broken. I just fixed it with commit https://github.com/libimobiledevice/libimobiledevice/commit/23069d10341ce637fdad7321d447c53752dba48c.
Make sure to remove invalid paring records by deleting the appropriate files or call ```idevicepair unpair``` for the affected devices.

ankitrwt, 2016-11-04:
I am having same problem on ubuntu 16.04 with Iphone6 ios 10.0.2
pair_record_generate_keys_and_certs(*pair_record, public_key) returns -5
i am using OpenSSL 1.0.2g 
[image]

ankitrwt, 2016-11-09:
using ubuntu 16.04  i successfully updated Iphone6 ios 10.1.1, stil as I mention in my above comment i am unable to get ideviceinfo it says ERROR: Could not connect to lockdownd, error code -5
Anyone have any Idea where I am doing wrong, and I think ios 10.0.2 have some issues as it was unable to upgrade iphone6 through idevicerestore

lifayk, 2016-12-11:
Same issue. iPhone 6S, IOS: 10.1.1 (14B100)
```
$ uname -a; cat /etc/issue; lsusb -d 05ac:12a8; apt list *imobile*; ideviceinfo 
Linux t4 4.8.0-28-generic #30-Ubuntu SMP Fri Nov 11 14:03:52 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 16.10 \n \l

Bus 002 Device 006: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Listing... Done
libimobiledevice-dev/yakkety 1.2.0+dfsg-3ubuntu1 amd64
libimobiledevice-doc/yakkety,yakkety 1.2.0+dfsg-3ubuntu1 all
libimobiledevice-utils/yakkety,now 1.2.0+dfsg-3ubuntu1 amd64 [installed]
libimobiledevice6/yakkety,now 1.2.0+dfsg-3ubuntu1 amd64 [installed,automatic]
libimobiledevice6-dbg/yakkety 1.2.0+dfsg-3ubuntu1 amd64
python-imobiledevice/yakkety 1.2.0+dfsg-3ubuntu1 amd64
GnuTLS error: Error in the pull function.
ERROR: Could not connect to lockdownd, error code -5
```

ankitrwt, 2016-12-11:
check SSL version you have and what SSL version libimobiledevice using
@lifayk

ankitrwt, 2016-12-12:
[https://github.com/libimobiledevice/libimobiledevice/issues/254](url)
@lifayk