libimobiledevice/libimobiledevice #300: Cloud ideviceinfo get some information about sim locked or carrier locked status?

jackczhang, 2016-03-31:
Cloud ideviceinfo get some information about sim locked or carrier locked status?
Just like get the info about find my iphone?

Kanong017, 2016-04-07:
sim卡锁的话,不知道你说的是那种,如果是网络锁,就只能靠建索引来判断,好像没有直接的方法
如果是sim卡的pin锁,可以得到

daniel-bluesea, 2017-05-04:
@jackczhang did you manage to understand @Kanong017 response?

nikias, 2017-05-04:
The only thing ideviceinfo can do is this:
```
ideviceinfo -k SIMStatus
```
If the device has a sim lock it should print ```kCTSIMSupportSIMStatusOperatorLocked```.
If the SIM is supported and ready to use it will print ```kCTSIMSupportSIMStatusReady```. Just check for yourself.

closed 2017-05-04 by nikias

daniel-bluesea, 2017-05-08:
SIMStatus returns the status of the currently inserted SIM card.
the carrier sim lock is a **device** status, can we get the device carrier sim lock status without the need of switching SIM cards?

nikias, 2017-05-08:
I don't know of any way to query this information, sorry.

daniel-bluesea, 2017-05-08:
some sites & forums claim that by using Apple GSX account and producing some kind of report you can get this (carrier sim lock) status along side with other device information.

some of this report information can be retrieved by using ideviceinfo, so i hoped that carrier sim lock status also can be retrieved by it..

nikias, 2017-05-08:
Still, AFAIK, there is no lockdown domain or key that would allow you to get this info. GSX is out of the scope of libimobiledevice.